RobotLocomotion/drake #5187: CMake error related to Director regarding Qt4

hammerheadCorvette, 2017-02-16:
New user here, forgive me if I'm missing something obvious. I am encountering a CMake error related to Director while attempting to build drake from source on Ubuntu 14.04. It seems as though I don't have Qt4 installed/configured properly:

```
CMake Error at /home/local/hammerhead/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find Qt4 (missing: QT_QTOPENGL_INCLUDE_DIR QT_QTOPENGL_LIBRARY)
  (found version "4.8.6")
Call Stack (most recent call first):
  /home/local/hammerhead/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /home/local/hammerhead/tools/cmake-3.5.2-Linux-x86_64/share/cmake-3.5/Modules/FindQt4.cmake:1333 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/dd-macros.cmake:2 (find_package)
  src/app/CMakeLists.txt:1 (setup_qt4)

-- Configuring incomplete, errors occurred!
See also "/home/local/hammerhead/Desktop/drake-distro/build/externals/director/CMakeFiles/CMakeOutput.log".
See also "/home/local/hammerhead/Desktop/drake-distro/build/externals/director/CMakeFiles/CMakeError.log".
make[2]: *** [director-prefix/src/director-stamp/director-configure] Error 1
make[1]: *** [CMakeFiles/director.dir/all] Error 2
make: *** [all] Error 2
```

As specified by the 14.04 specific instructions, I attempted to install the following dependencies; however, there was a libcheese and libcheese-gtk23 dependency issue (it seems):

```
$ sudo apt-get install --no-install-recommends \
>   libgl1-mesa-dri libqt4-dev libqt4-opengl-dev libqwt-dev \
>   libvtk-java libvtk5-qt4-dev python-lxml python-scipy python-vtk xvfb

Reading package lists... Done
Building dependency tree       
Reading state information... Done
libqt4-dev is already the newest version.
libqt4-dev set to manually installed.
python-lxml is already the newest version.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gnome-control-center : Depends: libcheese-gtk23 (>= 3.4.0) but it is not going to be installed
                        Depends: libcheese7 (>= 3.0.1) but it is not going to be installed
 unity-control-center : Depends: libcheese-gtk23 (>= 3.4.0) but it is not going to be installed
                        Depends: libcheese7 (>= 3.0.1) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

I attempted to independently sudo apt-get upgrade libcheese7 and libcheese-gtk23 but it looks like I do already have the latest versions. 

Any guidance would be greatly appreciated!

jamiesnape, 2017-02-16:
Try omitting `libgl1-mesa-dri` and `xvfb` and let me know if you get a different result, e.g.,
```
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install --no-install-recommends libqt4-dev libqt4-opengl-dev libqwt-dev libvtk-java libvtk5-qt4-dev python-lxml python-scipy python-vtk
```

jwnimmer-tri, 2017-02-16:
Related to #4947?

jamiesnape, 2017-02-16:
Quite probably.

hammerheadCorvette, 2017-02-17:
Thanks for the fix. This worked great - everything compiles successfully now.

jamiesnape, 2017-02-20:
Closed via #5207.

closed 2017-02-20 by jamiesnape