RobotLocomotion/drake #5190: Need a Way To Extract From the Whole-World Model a Model-Instance Model

liangfok, 2017-02-16:
# Problem Definition

The worlds we simulate will become increasingly complex in terms of the number of model instances present. Yet many controllers we create will still operate on individual model instances. To account for this discrepancy, there is an increasing need to support the ability to easily get a model of a portion of a world model. These "smaller" models can then be used by, for example, model-instance-specific-controllers.

# Proposed Action Item

Provide the method `RigidBodyTree::GetSubTree(int model_instance_id)` or `MultiBodyTree::GetSubtree(int model_instance_id)`. These methods are, in effect, a partial clone of the original "whole world" model.

amcastro-tri, 2017-02-16:
Thank you @liangfok. I added the reference to #5013. I will make sure we discuss this issue with the dynamics team next week in PA.
Would you point me to one of the latest application of this idea or capability? I know that for the Kuka arm the manipulation team was working with up to three RBT models loaded from three different urdf files. Is that in  master yet? probably by looking at the particular need we can think better about the form the an API for this would take.

siyuanfeng-tri, 2017-02-16:
this is a good to have feature, but not critical. 
@amcastro-tri it is in master. and i am making rbt from subtrees by using the stored model path in addition to the rbt. there is nothing wrong with it, just extra information to carry around, and extra parsing.

amcastro-tri, 2017-02-16:
Right, I am just trying to think about how this capability will be provided with MBT. Would you point me to the particular example you are thinking about? Pointing to a line number would be super helpful also :-)

siyuanfeng-tri, 2017-02-16:
https://github.com/RobotLocomotion/drake/blob/master/drake/examples/kuka_iiwa_arm/iiwa_world/controlled_iiwa_world_demo.cc#L133

it's not super obvious. Basically, some controller needs a RBT that only has the robot it's controller. So we can't pass the world tree (we should just call this RBforest or something like that), and have to make a new one from the model path, it' be cool to just clone only this part from the world tree.

amcastro-tri, 2017-02-16:
Thanks for the link! by the way, is that a typo in `CreateTreedFromFixedModelAtPose()`?, there is a `d` at the end of `Tree`.

siyuanfeng-tri, 2017-02-16:
its a double tree

jwnimmer-tri, 2018-12-21:
I suspect that this is already resolved in MBP, with our "model instances" while loading.  If not, we can re-open and/or file a more descriptive issue.

closed 2018-12-21 by jwnimmer-tri