RobotLocomotion/drake #5209: OSX Freezes When Launching Automotive Demos using automotive_demo.py

liangfok, 2017-02-17:
This issue was first mentioned in #5204. I've copy/pasted the original problem description below. I'm extracting it to be its own issue since I verified that it's an OSX-only issue. I wasn't able to replicate the freezing issue on Ubuntu 14.04.

# Problem Description

I recently encountered a problem where starting a demo using automotive_demo.py would basically cause my CPU to melt. The system would freeze for tens of seconds before displaying the various GUIs and by that time, the interesting part of the simulation was already over. This was on an OSX machine. It may be my fault since the demo is still undergoing review in (#5185).

jwnimmer-tri, 2017-02-17:
What command-line arguments?  The figure-eight demo usually burns CPU for sometimes tens of seconds before starting, especially in debug mode.

liangfok, 2017-02-17:
Both of the following commands exhibit the freezing behavior on my OSX machine (note that these targets are only in #5185 and are subject to change based on reviewer comments):

```
$ bazel run drake/automotive:demo_simple_car_and_trajectory_car_on_flat_terrain
$ bazel run drake/automotive:demo_dragway_with_trajectory_car
```

I'm using the default build mode (I simply ran `bazel build ...`). I'm not sure if that's Debug or Release. How do I tell?

jwnimmer-tri, 2017-02-17:
It's release by default, per `tools/bazel.rc`.  I guess I don't have an explanation for why the dragway is slow.

liangfok, 2017-02-17:
Using @rpoyner-tri's #5208 `--dry-run` option, I narrowed down the freezing issue to be due to the execution of `lcm-spy`:

```
$ sh -x drake/automotive/lcm-spy
+ eval ''
+ unset CLASSPATH
+ JVM_FLAGS_CMDLINE=()
++ uname -s
++ tr A-Z a-z
+ PLATFORM=darwin
+ ARGS=()
+ is_windows
+ [[ darwin =~ msys_nt* ]]
+ false
+ self=drake/automotive/lcm-spy
+ [[ drake/automotive/lcm-spy != /* ]]
+ self=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy
+ [[ '' != 1 ]]
+ [[ -z '' ]]
+ true
+ [[ -e /Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles ]]
+ JAVA_RUNFILES=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles
+ break
+ [[ -n /Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles ]]
+ export TEST_SRCDIR=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles
+ TEST_SRCDIR=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles
+ is_windows
+ [[ darwin =~ msys_nt* ]]
+ false
+ RUNFILES_MANIFEST_FILE=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles/MANIFEST
+ RUNFILES_MANIFEST_ONLY=%runfiles_manifest_only%
+ '[' -z %runfiles_manifest_only% ']'
+ JAVABIN=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles/local_jdk/bin/java
+ [[ '' == 1 ]]
+ [[ lcm.spy.Spy == \-\-\p\r\i\n\t\_\j\a\v\a\b\i\n ]]
+ [[ '' == 1 ]]
+ RUNPATH=/Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles/drake/
+ RUNPATH=drake/automotive/lcm-spy.runfiles/drake/
+ CLASSPATH=drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libautomotive-java.jar:drake/automotive/lcm-spy.runfiles/drake/../lcm/liblcm-java.jar:drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar:drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libviewer-java.jar
+ [[ -n '' ]]
+ [[ -n '' ]]
+ [[ -n '' ]]
+ ARGS=(${JVM_DEBUG_FLAGS} ${JVM_FLAGS} "${JVM_FLAGS_CMDLINE[@]}" ${MAIN_ADVICE} lcm.spy.Spy "${ARGS[@]}")
+ (( 305 > 120000 ))
+ exec /Users/liang/dev/drake-distro-2/bazel-bin/drake/automotive/lcm-spy.runfiles/local_jdk/bin/java -classpath drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libautomotive-java.jar:drake/automotive/lcm-spy.runfiles/drake/../lcm/liblcm-java.jar:drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar:drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libviewer-java.jar lcm.spy.Spy
Feb 17 13:55:56  java[17932] <Warning>: void CGSUpdateManager::log() const: conn 0x138ab: spurious update.
ClassDiscoverer: java.lang.NoClassDefFoundError: com/jidesoft/swing/RangeSlider
                 jar: drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar
                 class: info/monitorenter/gui/chart/controls/RangeChooserPanel.class
ClassDiscoverer: java.lang.NoClassDefFoundError: org/apache/xmlgraphics/java2d/ps/EPSDocumentGraphics2D
                 jar: drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar
                 class: info/monitorenter/gui/chart/events/Chart2DActionSaveEpsSingletonApacheFop$Chart2DActionSaveEps.class
Found 12 LCM types
Exception in thread "main" java.lang.NullPointerException
        at java.awt.Window.init(Window.java:497)
        at java.awt.Window.<init>(Window.java:537)
        at java.awt.Frame.<init>(Frame.java:420)
        at javax.swing.JFrame.<init>(JFrame.java:233)
        at lcm.spy.Spy.<init>(Spy.java:51)
        at lcm.spy.Spy.main(Spy.java:540)
```

@soonho-tri, can you check whether executing the following commands on your Macintosh or `macsim` results in the same freezing behavior? I'd like to know if this is only on my machine. Thanks!

```
cd drake-distro/bazel-bin
sh -x drake/automotive/lcm-spy
```

soonho-tri, 2017-02-17:
> @soonho-tri, can you check whether executing the following commands on your Macintosh or macsim results in the same freezing behavior? I'd like to know if this is only on my machine. Thanks!

I'm on it...

soonho-tri, 2017-02-17:
I have

[image]

```
$ sh -x drake/automotive/lcm-spy
+ eval ''
+ unset CLASSPATH
+ JVM_FLAGS_CMDLINE=()
++ uname -s
++ tr A-Z a-z
+ PLATFORM=darwin
+ ARGS=()
+ is_windows
+ [[ darwin =~ msys_nt* ]]
+ false
+ self=drake/automotive/lcm-spy
+ [[ drake/automotive/lcm-spy != /* ]]
+ self=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy
+ [[ '' != 1 ]]
+ [[ -z '' ]]
+ true
+ [[ -e /Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles ]]
+ JAVA_RUNFILES=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles
+ break
+ [[ -n /Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles ]]
+ export TEST_SRCDIR=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles
+ TEST_SRCDIR=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles
+ is_windows
+ [[ darwin =~ msys_nt* ]]
+ false
+ RUNFILES_MANIFEST_FILE=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles/MANIFEST
+ RUNFILES_MANIFEST_ONLY=%runfiles_manifest_only%
+ '[' -z %runfiles_manifest_only% ']'
+ JAVABIN=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles/local_jdk/bin/java
+ [[ '' == 1 ]]
+ [[ lcm.spy.Spy == \-\-\p\r\i\n\t\_\j\a\v\a\b\i\n ]]
+ [[ '' == 1 ]]
+ RUNPATH=/Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles/drake/
+ RUNPATH=drake/automotive/lcm-spy.runfiles/drake/
+ CLASSPATH=drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libautomotive-java.jar:drake/automotive/lcm-spy.runfiles/drake/../lcm/liblcm-java.jar:drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar:drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libviewer-java.jar
+ [[ -n '' ]]
+ [[ -n '' ]]
+ [[ -n '' ]]
+ ARGS=(${JVM_DEBUG_FLAGS} ${JVM_FLAGS} "${JVM_FLAGS_CMDLINE[@]}" ${MAIN_ADVICE} lcm.spy.Spy "${ARGS[@]}")
+ (( 305 > 120000 ))
+ exec /Users/soonhok/work/drake-distro/bazel-bin/drake/automotive/lcm-spy.runfiles/local_jdk/bin/java -classpath drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libautomotive-java.jar:drake/automotive/lcm-spy.runfiles/drake/../lcm/liblcm-java.jar:drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar:drake/automotive/lcm-spy.runfiles/drake/drake/lcmtypes/libviewer-java.jar lcm.spy.Spy
ClassDiscoverer: java.lang.NoClassDefFoundError: com/jidesoft/swing/RangeSlider
                 jar: drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar
                 class: info/monitorenter/gui/chart/controls/RangeChooserPanel.class
ClassDiscoverer: java.lang.NoClassDefFoundError: org/apache/xmlgraphics/java2d/ps/EPSDocumentGraphics2D
                 jar: drake/automotive/lcm-spy.runfiles/drake/../net_sf_jchart2d_jchart2d/jar/jchart2d-3.3.2.jar
                 class: info/monitorenter/gui/chart/events/Chart2DActionSaveEpsSingletonApacheFop$Chart2DActionSaveEps.class
Found 12 LCM types
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
java.net.SocketException: Can't assign requested address
```

liangfok, 2017-02-17:
Thanks @soonho-tri. It's good to know this problem is only occurring on my machine.

liangfok, 2017-02-19:
I updated and rebooted my OSX machine and the problem went away by itself. While I still don't know the underlying cause, the fact that it now works on my machine merits closing this issue.

closed 2017-02-19 by liangfok