RobotLocomotion/drake #5210: Certain Director Unit Tests Timeout Resulting in CI Failure

liangfok, 2017-02-17:
This is an issue for tracking the occurrence of these errors.

# Example Error

```
12:18:40       Start  1: test_testAffordancePanel
12:18:40 
12:18:40 1: Test command: /usr/bin/xvfb-run "-a" "-s" "-screen 0 1024x768x24" "/home/ubuntu/workspace/linux-clang-experimental-debug/build/install/bin/directorPython" "/home/ubuntu/workspace/linux-clang-experimental-debug/externals/director/src/python/tests/testAffordancePanel.py" "--testing" "--data-dir" "/home/ubuntu/workspace/linux-clang-experimental-debug/externals/director/../../../drc-testing-data" "--output-dir" "/home/ubuntu/workspace/linux-clang-experimental-debug/build/externals/director/Testing/Temporary" "--iiwa-drake"
12:18:40 1: Environment variables: 
12:18:40 1:  COVERAGE_PROCESS_START=/home/ubuntu/workspace/linux-clang-experimental-debug/build/externals/director/python-coverage/coverage.conf
12:58:40  1/22 Test  #1: test_testAffordancePanel ...........................***Timeout 2400.01 sec
```

# Example Logs
* https://drake-jenkins.csail.mit.edu/job/linux-clang-experimental-debug/3547/

patmarion, 2017-02-17:
My first guess is to question whether the test is hanging because of an issue with xvfb-run.  I would like to see CI start an Xvfb server and set the DISPLAY environment variable, then run tests without the xvfb-run wrapper.  Please re-assign this issue to someone who could perform that experiment on CI.

patmarion, 2017-02-17:
The first google hit for "xvfb-run hang" is here, notice that they report it hangs with run with the "-a" flag as is used on drake CI:

http://stackoverflow.com/questions/32630250/xvfb-run-hanging-on-server

I do not recommend xvfb-run -a for CI.

liangfok, 2017-02-17:
@david-german-tri you have access to the CI servers right? Feel free to re-delegate...

patmarion, 2017-02-17:
You can launch Xvfb server like this:

`Xvfb :99 -ac -screen 0 1280x1024x16 &`

or

```
/sbin/start-stop-daemon --start --quiet --pidfile /tmp/custom_xvfb_99.pid --make-pidfile \
--background --exec /usr/bin/Xvfb -- :99 -ac -screen 0 1280x1024x16
```

Then set environment variable `export DISPLAY=:99.0` before running tests.

david-german-tri, 2017-02-17:
@BetsyMcPhail Do you know why we're using `xvfb-run -a` on CI?  Any thoughts on @patmarion's recommendation above?

kunimatsu-tri, 2017-02-22:
Any updates on this? I'm asking because this seems very related to #5227 which will be blocking a PR to be merged.

patmarion, 2017-02-22:
I don't think your tests in #5227 are being run with xvfb-run.  Only director tests use xvfb-run as far as I know.

kunimatsu-tri, 2017-02-22:
Oh, I thought this would affect all the CI environment.

If no, I need the same change to my test's CI environment.

patmarion, 2017-02-22:
My recommendation to use Xvfb would allow all tests in the CI environment to run.  Using xvfb-run must be specified per-test.

kunimatsu-tri, 2017-02-22:
Just to be clear, my tests seems to be run with xvfb-run: https://drake-jenkins.csail.mit.edu/job/linux-xenial-clang-experimental/1725/consoleText

And, your recommendation sounds good to me.

patmarion, 2017-02-22:
in that log, i only see director tests using with xvfb-run.

kunimatsu-tri, 2017-02-22:
Ah, right. I was seeing director tests part.

So, my test doesn't have neither xvfb-run nor Xvfb for now.

david-german-tri, 2017-02-24:
Bumping priority since this is related to, and perhaps blocking, #5227.  @jamiesnape @BetsyMcPhail, what do you think about running Xvfb as a daemon on the CI instances, as @patmarion suggests above?  

https://github.com/RobotLocomotion/drake/issues/5210#issuecomment-280732532

david-german-tri, 2017-02-24:
If I'm understanding correctly, the Director tests will automagically stop using `xvfb-run` if `DISPLAY` is set: https://github.com/RobotLocomotion/director/blob/master/src/python/tests/CMakeLists.txt#L102

patmarion, 2017-02-24:
Yes.  Note that the check for DISPLAY happens at configure time.  I would eventually remove all the xvfb-run code if drake-ci decides to switch to Xvfb server daemon, or at least add an explicit option rather than checking for the DISPLAY environment variable.

jamiesnape, 2017-02-27:
The daemon is probably going to be fine. I need to check a couple of things before we go that route. Certainly, we would want to start/restart it for each CI run rather than when the instance starts in case something causes it to crash and burn on a previous CI run.

liangfok, 2017-03-01:
I just encountered this again in this build:
https://drake-jenkins.csail.mit.edu/job/linux-clang-experimental-debug/4027

Oddly, this time around, I see the "Unknown file extension in readPolyData" exception being thrown and yet the unit test continues to run until it times out. This exception was historically observed here: https://github.com/RobotLocomotion/drake/issues/1789#issuecomment-192880790.

```
08:34:06       Start  1: test_testAffordancePanel
08:34:06 
08:34:06 1: Test command: /usr/bin/xvfb-run "-a" "-s" "-screen 0 1024x768x24" "/home/ubuntu/workspace/linux-clang-experimental-debug/build/install/bin/directorPython" "/home/ubuntu/workspace/linux-clang-experimental-debug/externals/director/src/python/tests/testAffordancePanel.py" "--testing" "--data-dir" "/home/ubuntu/workspace/linux-clang-experimental-debug/externals/director/../../../drc-testing-data" "--output-dir" "/home/ubuntu/workspace/linux-clang-experimental-debug/build/externals/director/Testing/Temporary" "--iiwa-drake"
08:34:06 1: Environment variables: 
08:34:06 1:  COVERAGE_PROCESS_START=/home/ubuntu/workspace/linux-clang-experimental-debug/build/externals/director/python-coverage/coverage.conf
09:13:08 1: Traceback (most recent call last):
09:13:08 1:   File "/home/ubuntu/workspace/linux-clang-experimental-debug/build/install/lib/python2.7/dist-packages/director/consoleapp.py", line 51, in onStartup
09:13:08 1:     func()
09:13:08 1:   File "/home/ubuntu/workspace/linux-clang-experimental-debug/build/install/lib/python2.7/dist-packages/director/mainwindowapp.py", line 362, in loadData
09:13:08 1:     openDataHandler.openGeometry(filename)
09:13:08 1:   File "/home/ubuntu/workspace/linux-clang-experimental-debug/build/install/lib/python2.7/dist-packages/director/opendatahandler.py", line 39, in openGeometry
09:13:08 1:     polyData = ioUtils.readPolyData(filename)
09:13:08 1:   File "/home/ubuntu/workspace/linux-clang-experimental-debug/build/install/lib/python2.7/dist-packages/director/ioUtils.py", line 23, in readPolyData
09:13:08 1:     raise Exception('Unknown file extension in readPolyData: %s' % filename)
09:13:08 1: Exception: Unknown file extension in readPolyData: /home/ubuntu/workspace/linux-clang-experimental-debug/externals/director/../../../drc-testing-data
09:13:08 1: 
09:13:08 1: TESTING PROGRAM RETURNING EXIT CODE: 0
09:14:06  1/23 Test  #1: test_testAffordancePanel ...........................***Timeout 2400.02 sec
```

patmarion, 2017-03-01:
The printing of `Exception: Unknown file extension in readPolyData:` in some tests is harmless.  It began last month and I intend to fix it soon.  The exception is handled and printed for informational display.  The historical case you link to https://github.com/RobotLocomotion/drake/issues/1789#issuecomment-192880790 had a different cause.

I still believe the timeout is due to xvfb-run hanging.  Note that the test prints `TESTING PROGRAM RETURNING EXIT CODE: 0`.  That is the last thing printed before exit, and the return code 0 indicates a successful test.

liangfok, 2017-03-01:
Thanks for the analysis! I believe you're correct.

patmarion, 2017-03-01:
I see in that log that the timeout is on the very first test launched with xvfb-run.  Then the second test takes 70 seconds to run (it should only take 1s).  After that, things are ok.  So perhaps xvfb-run takes a long time to initialize the first time it is launched?

As a side note: there is a test that runs ik solve 128 times.  On a build with snopt enabled, this test runs in <2 seconds.  But I see the test took 10 minutes on drake CI (with ipopt i assume).  If that's a problem, some fixes i could propose are 1.  reduce the number of ik solves in this test, or 2. disable the test when snopt is not available, or 3. improve ipopt performance on ik problems

jonathan-owens, 2017-05-11:
Do we think [this](https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=586249) is related?

jamiesnape, 2017-05-11:
No, we do not yet run anything Director-related during the Bazel builds.

jamiesnape, 2017-11-15:
We no longer run Director unit tests.

closed 2017-11-15 by jamiesnape