RobotLocomotion/drake #5211: Document sdf/urdf collision semantics in Drake's contact filtering documentation

SeanCurtis-TRI, 2017-02-17:
PR #4890 provides documentation of how Drake handles collision filtering.  There are sections that incomplete.  Specifically, the discussion of the collision state of SDF and URDF files.  For example, by default is self-collision disabled for whole models?  What tags exist in each of the specifications which are *not* accounted for in this discussion.  How are nested collision statements resolved?  Etc.

EricCousineau-TRI, 2020-04-16:
Is this relevant any more?

SeanCurtis-TRI, 2020-04-16:
Just as relevant today as ever.

jwnimmer-tri, 2022-04-18:
FYI whatever happens for #14785 should also be reflected in the docs.

jwnimmer-tri, 2023-01-30:
See also #18350.

jwnimmer-tri, 2024-03-20:
I'm not entirely sure what was wanted here.  Possibly things like https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html#ada6a17fc10f55293cc507b583220f8d7 are good enough?

In any case, if there is still more to finish that's fine, but we need a much more specific inventory of missing docs than what was here.

closed 2024-03-20 by jwnimmer-tri