RobotLocomotion/drake #5216: Re-enable visualizer_startup_test.py

jwnimmer-tri, 2017-02-17:
The `drake/regtests/visualizer_startup_test.py` sayeth:
```
# TODO(patmarion)
# The visualization test is currently disabled due to a recent change in
# dependencies.  Now director depends on drake, so the drake-visualizer
# program is compiled after drake.  The ctest scripts running continuous
# integration will be updated to run director tests which will cover the
# visualizer.  Afterward, drake developers will decide whether we remove
# this test here or add new support for running tests after superbuild.
```

So, should we re-enable this test, or remove it?

david-german-tri, 2017-02-17:
I believe we should remove it.  Will PR and send to @patmarion for review.

patmarion, 2017-02-17:
yep, can remove.  lots of visualizer tests are now run via Director tests.

PR #5217 (2017-02-18): Remove drake/regtests.
@patmarion for feature review.

Fixes #5216.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5217)
closed 2017-02-18 by jwnimmer-tri