RobotLocomotion/drake #5238: support adding drake_cc_binary as a drake_cc_test with some command line flags

RussTedrake, 2017-02-21:
For a specific example in cmake, see
https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Pendulum/CMakeLists.txt#L56

I'd like to do the equivalent in bazel.  I've added a drake_cc_binary for pendulum_run_swing_up, but now want to add a test that uses the command line arguments to run faster than realtime (without having to write a one-off python wrapper).  

when asked quickly, @jwnimmer-tri said "We should share that shell script across BUILD files though I think, or wrap it up into a `cc_noncrash_test` macro if we wanted to do that."

jwnimmer-tri, 2017-02-21:
I suspect that a small macro around https://bazel.build/versions/master/docs/be/shell.html#sh_test with appropriate `args` based on the `$(location some_cc_binary)` would do the trick.

jwnimmer-tri, 2017-02-21:
Oops, https://bazel.build/versions/master/docs/be/c-cpp.html#cc_test does support `args`; I wonder how I missed that the first time?  I just tested that with a kuka simulation demo and it worked.

So the only request here is to add a `drake_cc_test` wrapper, similarly to `drake_cc_googletest`.

david-german-tri, 2017-03-03:
I am confused - #5266 contradicts the last comment.

jwnimmer-tri, 2017-03-03:
My bad.  Let's burn the sh_tests with fire.  Sorry!

SeanCurtis-TRI, 2017-03-03:
@liangfok had recommended this approach as a way to make sure examples get run regularly to make sure they don't become detached from Drake.  If this is trashed, what is the alternative?  Or are we simply not doing this?

SeanCurtis-TRI, 2017-03-03:
So, I *cannot* just execute another binary in a text context? 

This one requires that the file be located in a `test` sub-directory.  However, if I move it I can replace the `drake_cc_binary` with `drake_cc_test`; the executable will get built once, and in one of those contexts it'll be included in the tests.  Is that about right?

jwnimmer-tri, 2017-03-03:
Stay tuned; we think just publishing an example (fixing this issue) will clarify things immensely.

PR #5266 (2017-02-23): Add pendulum_run_swing_up_test to Bazel
Fixes #5238.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5266)
PR #5402 (2017-03-04): Replace exec.sh with ordinary cc_test rules using args.
Fixes #5238.

@SeanCurtis-TRI for feature review, @jwnimmer-tri for platform review.

To see in action:

`bazel run //drake/examples/Pendulum:pendulum_run_swing_up`

`bazel test //drake/examples/Pendulum:pendulum_run_swing_up_test`

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5402)
closed 2017-03-04 by jwnimmer-tri