RobotLocomotion/drake #5263: Mysterious CI Failure: Numerous Unit Tests Timing Out For No Apparent Reason

liangfok, 2017-02-22:
# Example Problem

```
10:53:04 97% tests passed, 8 tests failed out of 265
10:53:04 
10:53:04 The following tests FAILED:
10:53:04 	188 - rotary_encoders_test (Timeout)
10:53:04 	189 - image_test (Timeout)
10:53:04 	190 - camera_info_test (Timeout)
10:53:04 	191 - depth_sensor_test (Timeout)
10:53:04 	192 - rgbd_camera_test (Timeout)
10:53:04 	193 - vtk_util_test (Timeout)
10:53:04 	194 - gyroscope_test (Timeout)
10:53:04 	195 - accelerometer_test (Timeout)
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/job/linux-clang-continuous-ros-debug/1232/

david-german-tri, 2017-02-22:
Well, that is weird.  `ctest` is reporting a timeout, but in fact, the logs show that those tests ran and passed.  For instance:

```
189: Running main() from gtest_main.cc
189: [==========] Running 8 tests from 1 test case.
189: [----------] Global test environment set-up.
189: [----------] 8 tests from TestImage
189: [ RUN      ] TestImage.EmptyTest
189: [       OK ] TestImage.EmptyTest (0 ms)
189: [ RUN      ] TestImage.InstantiateTest
189: [       OK ] TestImage.InstantiateTest (1 ms)
189: [ RUN      ] TestImage.InitializeAndAccessToPixelValuesTest
189: [       OK ] TestImage.InitializeAndAccessToPixelValuesTest (114 ms)
189: [ RUN      ] TestImage.CopyConstructorTest
189: [       OK ] TestImage.CopyConstructorTest (185 ms)
189: [ RUN      ] TestImage.AssignmentOperatorTest
189: [       OK ] TestImage.AssignmentOperatorTest (86 ms)
189: [ RUN      ] TestImage.MoveConstructorTest
189: [       OK ] TestImage.MoveConstructorTest (57 ms)
189: [ RUN      ] TestImage.MoveAssignmentOperatorTest
189: [       OK ] TestImage.MoveAssignmentOperatorTest (56 ms)
189: [ RUN      ] TestImage.ResizeTest
189: [       OK ] TestImage.ResizeTest (0 ms)
189: [----------] 8 tests from TestImage (499 ms total)
189: 
189: [----------] Global test environment tear-down
189: [==========] 8 tests from 1 test case ran. (499 ms total)
189: [  PASSED  ] 8 tests.
```

This smells like a `ctest` bug to me.  @jamiesnape @BetsyMcPhail any thoughts?

liangfok, 2017-02-23:
@soonho-tri reports:

There are unstable builds due to timeout at
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/linux-clang-continuous-ros-debug 

Build 1232

```10:53:04     188 - rotary_encoders_test (Timeout)
10:53:04     189 - image_test (Timeout)
10:53:04     190 - camera_info_test (Timeout)
10:53:04     191 - depth_sensor_test (Timeout)
10:53:04     192 - rgbd_camera_test (Timeout)
10:53:04     193 - vtk_util_test (Timeout)
10:53:04     194 - gyroscope_test (Timeout)
10:53:04     195 - accelerometer_test (Timeout)
```

Build 1235

```16:26:30     189 - rotary_encoders_test (Timeout)
16:26:30     190 - image_test (Timeout)
16:26:30     191 - camera_info_test (Timeout)
16:26:30     192 - depth_sensor_test (Timeout)
16:26:30     193 - rgbd_camera_test (Timeout)
```

Build 1237

```18:29:07     199 - car_sim_lcm (Timeout)```

jamiesnape, 2017-02-23:
> This smells like a ctest bug to me. @jamiesnape @BetsyMcPhail any thoughts?

I would be surprised. All `ctest` is doing is waiting for the process to return. If the process does not return in 60 seconds, it kills it, and reports a timeout.

liangfok, 2017-02-23:
I'm running the unit tests using Drake + ROS on my workstation using 0c7a92b7a1d484991a489e304e1cd9bda2ad3163.

# Attempt 1 - One unit test timed out

```
192/251 Test #192: car_sim_lcm ........................................***Timeout  60.01 sec
...
99% tests passed, 1 tests failed out of 251

Label Time Summary:
medium    = 396.40 sec (10 tests)
small     = 315.38 sec (241 tests)

Total Test time (real) = 711.90 sec

The following tests FAILED:
	192 - car_sim_lcm (Timeout)
```

liangfok, 2017-02-23:
Based on the results above, it's pretty clear to me that running `car_sim_lcm` in debug mode is problematic. It's probably caused by the new contact modeling in #5197. We should either increase the timeout or disable this unit test in debug mode.

soonho-tri, 2017-02-23:
@liangfok , thank you for doing this! I still don't understand if this is ROS-specific or not. Do you have an idea?

soonho-tri, 2017-02-23:
Paging @SeanCurtis-TRI , the author of #5197.

liangfok, 2017-02-23:
It's possible that running the unit tests through a ROS environment somehow slows down the unit test. I'm trying to figure out exactly what commands are being executed by the CI driver. For example, one thing I'm looking into is whether the unit tests are being run directly through `ctest` or indirectly through `rostest`.

liangfok, 2017-02-23:
Based on the link below, I believe `ctest` is being run directly. Thus, I'm not sure why these timeouts would be ROS specific.

https://github.com/RobotLocomotion/drake-ci/blob/e4e8fd71a7bcdcdf83ec70a649e13353d6de9541/driver/configurations/catkin/step-drake-tests.cmake#L11

SeanCurtis-TRI, 2017-02-23:
I can make two definitive statements relating to `car_sim_lcm`.

1) Due to the new model, we swapped from the default RK2 integrator (with a timestep of 1e-3) to the Semi-explicit Euler with a timestep of 1e-4.  So, best case scenario, it should run at 1/10th the speed.
2) I avoided running it in debug mode like the plague. 

It's conceivable that it's generally close to failing everywhere, but ROS provides just enough of a sliver to push it over the edge?

liangfok, 2017-02-23:
> It's conceivable that it's generally close to failing everywhere, but ROS provides just enough of a sliver to push it over the edge?

Good question. It's possible. I will do more testing to see exactly how long it takes to run to see how close we are to the 60s limit.

soonho-tri, 2017-02-23:
/cc @kunimatsu-tri

liangfok, 2017-02-23:
Here's one initial result that indicates the `car_sim_lcm` unit test takes one minute and 44.61 seconds to run in debug mode:
```
$ /usr/bin/time --format=%E,%U,%M  ./automotive/car_sim_lcm --simulation_sec 0.1
...
1:44.61,104.27,96544
```

soonho-tri, 2017-02-23:
@liangfok , good to know. I think a solution for this issue is to increase test timeout for `car_sim_lcm` to 120sec (2min)?

kunimatsu-tri, 2017-02-23:
I created a PR as a temporal solution just in case, but it's OK for me to close it if @liangfok found the final solution.

soonho-tri, 2017-02-23:
@kunimatsu-tri , sounds good to me. Let's see how CI works with that PR.

BTW, could you mention the PR in this issue so that Github links the two?

kunimatsu-tri, 2017-02-23:
Sure. #5294

soonho-tri, 2017-02-23:
Ooops. It was already displayed up above. I missed that. Thank you @kunimatsu-tri !

jwnimmer-tri, 2017-02-23:
How about change `simulation_sec` from `0.1` to `0.001`?  Hopefully that being the test to a tolerable time when run automatically.

kunimatsu-tri, 2017-02-23:
Yeah, that's another possible solution.

@liangfok: Is this OK for you?

liangfok, 2017-02-23:
Yes I believe that will work. I will verify then submit a PR.

jwnimmer-tri, 2017-02-23:
Now that #5266 has merged, the test is in Bazel, so the BUILD file also needs the shorter duration.

kunimatsu-tri, 2017-02-23:
> Yes I believe that will work. I will verify then submit a PR.

Great. 
If that won't take much time, just ignore my feature review request in #5294.

liangfok, 2017-02-24:
Sorry for the delay. I had to switch computers and ran into #5298.

The unit test takes 1:16 on my home desktop PC:

```
$ cd ~/dev/drake_catkin_workspace/build/drake/drake
$ /usr/bin/time --format=%E,%U,%M  ./automotive/car_sim_lcm --simulation_sec 0.1
...
1:16.92,76.66,96504
```

Decreasing the simulation duration by 10X gives:

```
$ /usr/bin/time --format=%E,%U,%M  ./automotive/car_sim_lcm --simulation_sec 0.01
...
0:07.85,7.79,96784
```
Since I believe the unit test's efficacy is not decreased by this change, I created #5299.

PR #5299 (2017-02-24): Decreased car_sim_lcm unit test runtime by 10X to 0.01s.
Potentially resolves #5263.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5299)
closed 2017-02-24 by jwnimmer-tri