RobotLocomotion/drake #5267: Need multi-arm iiwa model for bimanual manipulation planning

sammy-tri, 2017-02-22:
To start doing tests with bimanual manipulation, we'll need to have a single model (well, single RigidBodyTree however we get there) with both arms for model-based planning.

liangfok, 2017-02-22:
My first spike test is to simply load the existing URDF of the IIWA robot into the same `RigidBodyTree` twice. I will then instantiate two separate PID controllers, one for each IIWA model instance.

naveenoid, 2017-02-22:
I suspect this is the only way to go, and by creating a wrapping over the resulting diagram..the practical system still requires independent implementation in anycase...

patmarion, 2017-02-24:
why not a single urdf with the full two armed two grippered robot in it?

liangfok, 2017-02-24:
I'm hesitant to cause an explosion in the number of URDFs that then need to be kept in sync. I supposed one option is to use `xacro`.

liangfok, 2017-02-24:
We may need to address #4326.

patmarion, 2017-02-24:
what's wrong with having urdf files?  they would be generated from templates, clearly identified as generated, and checked into the repo.  otherwise you will continue to have hard coded values in the kuka sim c++ files, which i'd argue is currently unusable by external programs like controllers and visualizers.

liangfok, 2017-02-24:
I just want to ensure all URDFs that include an IIWA model are kept in sync. I agree auto-generating the URDFs from templates is a good solution and will look into whether `xacro` can serve this purpose. One option might be to provide users with instructions on which version of `xacro` to use.

patmarion, 2017-02-24:
i think the two arm version ought to be different, so that link names can be like `left_arm_link_1` etc

liangfok, 2017-02-24:
Oh yeah, you just reminded me that URDF does not support more than one model instance! I would argue that this is one reason not to save more than one instance of the IIIWA in a single URDF. Perhaps we should go with SDF, which does support including multiple models in a single file.

liangfok, 2017-02-27:
I now know a working SDF syntax for including multiple URDF model instances into a world: #5332. I suspect, however, that updating Drake's parsers to support this feature will take a significant amount of time since it will require a resolution to #5317. To more quickly resolve this issue, my plan is to programmatically create the `RigidBodyTree` with two iiwa manipulators. Once #5317 is resolved, we can switch to the create-the-world-before-tree-creation strategy.

liangfok, 2017-04-05:
With the landing of #5636, we can easily generate URDFs containing any number of iiwa arms, as exemplified by the dual-arm model introduced by #5660. The next logical step is to add support for the inclusion of end effectors and furniture like the extra heavy duty table. These models, however, are in SDF format, meaning they cannot be added to a URDF via xacro.

@patmarion, you're using a manually-created URDF containing both an iiwa and a gripper with Director right? If so, did you convert the gripper's SDF into URDF so that it can be included with the iiwa model?

patmarion, 2017-04-05:
The urdf files I used are in the `spartan` repo.  I used a wsg50 urdf and meshes that I got from @gizatt from his earlier work with that gripper and the gelsight sensor.  I believe the got the urdf from the original upstream source here:

http://wiki.ros.org/wsg50
https://github.com/nalt/wsg50-ros-pkg/tree/master/wsg_50_simulation

liangfok, 2017-04-05:
Thanks! I will look into using that URDF. Thankfully it's [released under a BSD license](https://github.com/nalt/wsg50-ros-pkg/blob/a3ab07374a6cff13a90af3c23778622a71d1db14/wsg_50_driver/package.xml#L16).

jwnimmer-tri, 2018-06-09:
@sammy-tri Is this issue resolved to our satisfaction?

closed 2018-06-11 by sammy-tri