RobotLocomotion/drake #5268: Segfaults and std::abort make the Python interface difficult to use

rdeits, 2017-02-22:
This isn't a bug, strictly, but it's a design choice that's been making my life difficult, so I figured I should at least mention it. The policy inside drake seems to be to use `DRAKE_DEMAND` to do things like check if function inputs are sane. Since `DRAKE_DEMAND` is a no-op in Release mode, doing something mildly stupid, even from Python, will segfault and crash the entire Python interpreter. In Debug mode, a failed `DRAKE_DEMAND` causes a `std::abort`, which *also* crashes the entire Python interpreter. Normal C++ exceptions would, on the other hand, get gracefully propagated up to Python and produce useful Python errors without crashing the interpreter itself. 

As an example, here are a couple of stupid things I've done in the last 24 hours that, rather than throwing an error, just caused all of Python to crash:

* trying to call `MathematicalProgram::GetSolution(x)` on a variable which belonged to the wrong model
* trying to call `MathematicalProgram::AddQuadraticConstraint(Q, q, x)` where the size of `Q` and `x` did not match

These are pretty common errors, and causing Python to crash (rather than error) even in Debug mode makes it much harder to diagnose them when they occur. This will only get worse as I try to write more Python code that calls Drake functions. 

So, the question is, why `std::abort` instead of throwing an exception? Is there any hope of making this more user-friendly?

jwnimmer-tri, 2017-02-22:
FYI/ nit: `DRAKE_ASSERT`s are turned off in release, but `DRAKE_DEMAND` will always be armed.

rdeits, 2017-02-22:
Interesting. Then I'm not sure what was going on in my first example, since calling `GetSolution(x)` on a missing variable `x` causes a segfault in Release mode and an abort in Debug mode, even though it's using `DRAKE_DEMAND` internally.

rdeits, 2017-02-22:
An example of my first case is:

```python
import pydrake.solvers.mathematicalprogram as mp
p1 = mp.MathematicalProgram()
x = p1.NewContinuousVariables(1, "x")
p2 = mp.MathematicalProgram()
p2.GetSolution(x)
```

which crashes with `SIGSEGV` in Release mode.

jwnimmer-tri, 2017-02-22:
So #1935 is the long-ago discussion on turning off correctness checks to get better runtime performance.  I believe that the Drake's mantra is "turn on assertions when developing new code", so that speaks to the ASSERT vs DEMAND coverage challenges expressed above.  As part of resolving that discussion, I added a way to arm the ASSERTs independently from CMake's Debug mode / `NDEBUG` / Eigen assertions / `-O0` / etc., so that might be one thing that helps here.

I think the bigger discussion is `throw` vs `abort` and when each should be used.  Having Python code wrapping Drake introduces new kinds of uses that merit more thought.  I will continue pondering.

Even aside from the big-picture discussions though, I think there's probably lots of low-hanging fruit to improve the diagnostics even within the current architecture and tools.  For any particular MathematicalProgram error conditions that are being hit often and give bad diagnostics, are there easy things we can do to repair them?

hongkai-dai, 2017-02-22:
I think I can change many of the `DRAKE_DEMAND` in `MathematicalProgram` to throw `std::runtime_error`. Will that solve this problem?

rdeits, 2017-02-22:
@jwnimmer-tri thanks for the pointer to #1935; that was helpful to read. As @hongkai-dai suggested, throwing `std::runtime_error` would make the exceptions easier to understand from Python (and would stop the interpreter itself from crashing), but I realize that might be incompatible with the eventual goal of using MathematicalProgram inside a controller's inner loop. The simplest change that would make my life better would be to throw a `std::runtime_error` instead of a `std::abort` in the various `drake_assert` macros.

jwnimmer-tri, 2017-02-22:
I don't think we would get buy-in to permanently change DRAKE_ASSERT / DRAKE_DEMAND action from `abort()` to `throw ...`; it's a well-understood mnemonic that the program will halt when using those tools.  I think the furthest that we could find consensus on is to have an optional hook where users could replace the `Abort` fail-action (e.g., with a function that throws) -- but I'm honestly not sure what the team would say about that.

If there are `MathematicalProgram` sanity checks that users are able to trip (e.g., not internal invariants) and those can turn into `DRAKE_THROW_UNLESS` and still execute quickly enough, then `DRAKE_THROW_UNLESS` is the best answer.  If we want both error-reporting using exceptions but to disable it in Release builds, then that is possible but we will probably need new sugar for it -- right now you'd have to do something like `#ifdef DRAKE_ASSERT_IS_ARMED \n DRAKE_THROW_UNLESS(...) \n #endif` which is a bit of a mouthful.  And unit testing it would be difficult (since it doesn't trip in all builds).

liangfok, 2017-02-23:
Possibly related historical discussion: https://github.com/RobotLocomotion/drake/pull/4728/files

rdeits, 2017-02-23:
Thanks @liangfok . I agree that throwing an abort makes sense when trying to catch bugs internal to drake, but throwing them from user-facing methods does not.

jwnimmer-tri, 2017-03-11:
When we last talked about this offline, there was general willingness to turn user-caused errors into thrown exceptions.  So if we find instances of that which are troublesome, I think PRs to adjust them will be okay.

There was also openness to adding a hook in the `DRAKE_ABORT` macro that would let _downstream projects that compile Drake_ choose a different action than `drake::Abort`.  We would forbid setting that hook in Drake's own C++ code, but would leave it open as a toggle a user can set.

hongkai-dai, 2017-03-11:
I think I have changed some of the user-caused error to throw exceptions. @Robin if you think there are more cases that should be changed to exceptions, please let me know, I will send a PR then.

RussTedrake, 2017-04-29:
I've got a WIP branch making python bindings for system classes.  (I'm trying to prototype optimizations on systems in python before pushing them to c++... i think that's a very important workflow).  This issue is going to come up again in spades.  

Is there a tangible plan for putting in a hook for python bindings to change the behavoir of DRAKE_DEMAND to throw instead of abort?  Changing DRAKE_DEMAND's to DRAKE_THROW_UNLESS in the core system seems like more than we intend.

jwnimmer-tri, 2017-05-18:
I am going to put this relative high up on my TODO list.

jwnimmer-tri, 2017-05-19:
WIP is at https://github.com/jwnimmer-tri/drake/tree/assertion-exception.  Should be about 3 PRs deep by the time its ready.

jwnimmer-tri, 2017-05-24:
So the objection in #6167 review is (rephrased) that Drake doesn't offer even the Basic exception safety guarantee, and moreover even if it did for its `throw` statements, magically having `DRAKE_DEMAND` turn into a `throw` instead of `abort` could defeat whatever basic exception safety was already in place and tested.  Without that kind of guarantee, isn't not clear that we're doing our users a favor by allowing execution to continue.

One incremental answer is to turn `DRAKE_DEMAND` failure into a pythonic `SystemExit` exception.  We'd get a better error message, at least.  The program would probably exit (unless the programmer decided to catch SystemExit and continue, in which case they get what they deserve), which mostly resolves questions about lacking Basic exception safety guarantee.

A more difficult answer is to offer the Basic exception safety guarantee in the core portions of Drake that get wrapped into python bindings.  This is probably also helpful to C++ library users directly.  We'd need to do a re-review of those modules first, and then enforce proper testing and code review for them moving forward.  This will be more expensive, but is probably required for Drake's endgame anyway.

I guess the thing I'm not 100% sure about is exactly how Python users want to use Drake right now.  If there's a small prototype Python program that gets edited run repeatedly (in a new interpreter every time), then having it SystemExit when there's a typo isn't a huge deal -- just re-run it.  If you're using a long-lived interactive interpreter, however, maybe that's too annoying?

/CC @ggould-tri

ggould-tri, 2017-05-24:
That's a solid summary of my objections; I also considered that incremental answer but don't know if it is practical.

jwnimmer-tri, 2017-05-24:
I wasn't sure either, but looking now at http://pybind11.readthedocs.io/en/master/advanced/exceptions.html it seems likely that we could map `drake::assertion_error` into `SystemExit` easily.  Given the meaningful improvement in usability, I might try that first now while we iterate on setting a future goal.

rdeits, 2017-05-24:
My experience is that in general when doing exploration and research, I typically have relatively long-lived interpreter sessions with some important or expensive state (like a dataset that I'm in the process of collecting or a model that I'm designing). So a python module that raises SystemExit on user error would hamper that workflow. For reference, I can't think of any python module I've used that would intentionally cause the interpreter to exit. 

Throwing SystemExit is better than just aborting with no backtrace, though. So that change is an improvement, but not enough to make Drake behave like other python libraries.

jwnimmer-tri, 2017-05-24:
Thanks; that was my guess and my own experience sometimes as well.  I also agree that SystemExit is extremely unusual from a module.

PR #6175 will be a first step to (hopefully) help for now.  Once I can gather a set of relevant folks, I'll try to map out a better future plan.

jwnimmer-tri, 2017-06-09:
Update -- with events and travel its been hard to gather a technical forum of developers to agree on an approach here.  It's still a high priority, but delayed.

jwnimmer-tri, 2018-02-03:
We have quorum to pick this up again.  I'm working up a proposal to bound off a few folks at TRI, then will publish here likely within a week.

RussTedrake, 2018-03-04:
@jwnimmer-tri — was there a resolution on this?  My experience with pydrake in jupyter has been pleasant enough that perhaps this has been resolved?

EricCousineau-TRI, 2018-12-10:
There have not been any complaints on this; there is some more discussion on the usage of `DRAKE_ABORT`, but I believe that can be tackled as a separate issue.

jwnimmer-tri, 2018-12-10:
There is still more do to here.

jwnimmer-tri, 2019-03-22:
Status on current master:
- DRAKE_THROW_UNLESS raises an exception (in both C++ and Python).
- DRAKE_DEMAND (and DRAKE_ASSERT when armed) abort in C++ and raise SystemExit in Python.
- DRAKE_ABORT and DRAKE_ABORT_MSG are gone (well, deprecated and unused by Drake; soon to be withdrawn from our headers).
- DRAKE_UNREACHABLE aborts in both C++ and Python.  However, absent extreme action by the user (such as casting an `int` value to an `enum` value that is not one of the named values), it should never occur in practice.

There is more work to do on making Drake's use of the macros more consistent and adding better error details (#7827), but in terms of the "python interpreter hard exit", I think we are in pretty good shape now.

If anyone experiences segfaults from pydrake, please do open an issue for that particular case, so that we can be sure to add more fail-fast checking to whatever API allowed the user to cause a segfault.

closed 2019-03-22 by jwnimmer-tri