RobotLocomotion/drake #5280: drake-visualize segfaults on ubuntu 16.10 and 17.04

maidenone, 2017-02-23:
I have seen a QT linking issue in both 16.10 built with ninja and 17.04 built with make,
In both cases the system had vtk6 installed and vtk5 was checked out and built during the drake build process.

gdb crash output
```
Thread 1 "drake-visualize" received signal SIGSEGV, Segmentation fault.
0x00007fff6d2cd387 in QMetaObject::className() const () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
```

output from ldd drake-visualizer | grep libQt
```
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007fcfbf176000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007fcfbec82000)
	libQtOpenGL.so.4 => /usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4 (0x00007fcfbe019000)
	libQtPropertyBrowser.so => /home/maiden/Projects/drake-distro/build/install/lib/libQtPropertyBrowser.so (0x00007fcfbdd26000)
	libQtWebKit.so.4 => /usr/lib/x86_64-linux-gnu/libQtWebKit.so.4 (0x00007fcfb28af000)
	libQtSql.so.4 => /usr/lib/x86_64-linux-gnu/libQtSql.so.4 (0x00007fcfb266e000)
	libQtNetwork.so.4 => /usr/lib/x86_64-linux-gnu/libQtNetwork.so.4 (0x00007fcfb2319000)
```

I have checked that "/usr/lib/x86_64-linux-gnu/libQtCore.so.4" actually exists so it should not link against libQt5Core as the ldd output states libQtCore.. strange

Building with debug and running gdb gives:

```
Starting program: /home/maiden/Projects/drake-distro/build/install/bin/drake-visualizer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffd16c3700 (LWP 15067)]
[New Thread 0x7fffd0ec2700 (LWP 15068)]
[New Thread 0x7fffcbfff700 (LWP 15069)]
[New Thread 0x7fffbffff700 (LWP 15070)]
[New Thread 0x7fffbf7fe700 (LWP 15071)]
[Thread 0x7fffbffff700 (LWP 15070) exited]

Thread 1 "drake-visualize" received signal SIGSEGV, Segmentation fault.
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:294
294	../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: No such file or directory.
```

drake on 16.10 is built from git commit: a814c2fabb2fffbb2f09eb97609472b40899c933
drake on 17.04 is built from git commit: d96b2c113b083a2a5bdb2848369eb142f47b0165

maidenone, 2017-02-23:
Added a valgrind trace:
[valgrindDrake.txt](https://github.com/RobotLocomotion/drake/files/796436/valgrindDrake.txt)

patmarion, 2017-02-23:
Hi, it looks like the linkage for drake-visualizer is using Qt4, as it should.  The issue probably occurs in a dlopen, probably when drake-visualizer imports a python module.  My best guess is that vtk5 is not in your python path, so when `drake-visualizer` does a `import vtk` it gets the system installed vtk6, which is wrong, and it may be linked with qt5.

There are some instructions here which show how to prepend the vtk5 python directory to your python path (PYTHONPATH).  I'm not sure whether you built drake as a catkin package as is described in the instructions here, but that doesn't matter, just set PYTHONPATH to include the required vtk5 dirs (which are inside your drake build dir)

http://drake.mit.edu/from_source_ros_kinetic.html#step-6-modify-environment-variables-to-allow-vtk5-libraries-to-be-found

maidenone, 2017-02-23:
Booo! i already had that export, but found typos in my path. me be bad code monkey.. 
Thanks! now it works fine. we can close this ticket. We should maybe add something in the wiki about my experience. Works fine  now on 16.10 and 17.04!

patmarion, 2017-02-24:
Awesome, glad it's working now.  Sorry the import failure doesn't print a more helpful error message, that could be improved.

patmarion, 2017-02-24:
/cc @liangfok in case you want to put a link to this in the drake install instructions.

PR #5925 (2017-04-24): Adds FAQ entry explaining why drake-visualizer segfaults upon start on systems with VTK6.
Resolves #5280.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5925)
closed 2017-02-24 by patmarion