RobotLocomotion/drake #5288: Add an IDM-Controlled SimpleCar and TrajectoryCar on Dragway Demo

liangfok, 2017-02-23:
We want to demonstrate how an IDM-controlled `SimpleCar` can travel down a one-lane Dragway and avoid colliding with a `TrajectoryCar`. Here is the system diagram:

<pre>
               -------------    -------------------
               |           |    |                 |
    ---------->| SimpleCar |    |  TrajectoryCar  |
    |          |           |    |                 |
    |          -------------    -------------------
    |                |                   |
    |                |  state            |  state
    |                |                   |
    |                V                   V
    |          -----------------------------------
    |          |                                 |
    |          |         PoseAggregator          |
    |          |                                 |
    |          -----------------------------------
    |                          |
    |                          |  PoseBundle
    |                          |
    |                          V
    |                  ----------------
    |                  |              |
    |                  |  IdmPlanner  |
    |                  |              |
    |                  ----------------
    |                          |
    |                          |
    |   throttle / steering    |
    ----------------------------
</pre>

jwnimmer-tri, 2017-02-23:
https://github.com/RobotLocomotion/drake/blob/master/drake/automotive/idm_planner.h doesn't take a PoseBundle, it takes a one-dimensional x and xdot, for each of the two vehicles.

liangfok, 2017-02-24:
I suppose we have two options:

1. Modify `IdmPlanner` to take as input a `PoseBundle`.

2. Develop a new system that takes as input a `PoseBundle` and outputs an `x` and `xdot` for the vehicle under its control.

I thought option 1 was what we decided to do today, but option 2 is probably a better design from a software design perspective since it achieves better separation of concerns. Thoughts?

jwnimmer-tri, 2017-02-24:
Like I said in the meeting, I will write down the list of models and we will take it from there.  I should have it done by midnight, just be patient please.

jwnimmer-tri, 2017-02-24:
If we want to do this demo quickly, we can extract the IDM equation from `idm_planner.h` into a function (taking as arguments ego velocity, closing velocity, and longitudinal separation), and then wire up a custom System that takes 2x SimpleCarState as inputs and publishes DrivingCommand as output, using the IDM equation.  Then we just hook that up in the obvious way to SimpleCar and TrajectoryCar, and we'll have a working demo.

Its possible though that our time would be better invested in creating some of the other models outlined in the document, since the above-mentioned System would be relatively short-lived.

liangfok, 2017-04-12:
I am unassigning myself since the MaliputRailcar version is in #5756 and @jadecastro will PR the `SimpleCar` version.