RobotLocomotion/drake #5297: Constraint methods in System need revision

sherm1, 2017-02-24:
Per discussion with @edrumwri, the constraint methods that slipped into System before I saw them need revision. A few issues:
- These should be Calc methods, not Eval; Eval is for caching only.
- Position, velocity, and acceleration errors should be distinguishable. For example, an Assembly analysis or IK needs to drive the position constraints to zero but has no interested in v or a.
- Documentation is needed to specify what set of equations is being modeled. For example, are position constraint (holonomic) derivatives included in the velocity errors along with velocity constraints (nonholonomic)?
- These methods should be templatized by T, they shouldn't be doubles.
- The velocity change method should be removed or replaced with one that is not specific to mechanical systems and does not require formation of an n^2 matrix; should permit an O(n) implementation.
- The error norm is mixing units so should involve a scaling matrix but I don't think there is one. I'm not clear why this is a System API method since the constraint errors are already available. Choice of norm is usually made be a solver; could use a utility method that scales?

RussTedrake, 2017-02-24:
fwiw -- http://drake.mit.edu/doxygen_matlab/class_manipulator.html#ae9a5cd0ce75ce832f0070c87c894d6fe
there are a number of constraint implementations (e.g. constant-length cable drive) in matlab that we can move over when this is ready.

RussTedrake, 2017-02-24:
another important distinction is whether constraints are equality constraints or inequality constraints.  of course, with our constraint classes you can simply ask the constraint about it's bounds; in matlab we kept separate lists of constraints for the two distinct types because they are treated very differently in the dynamics code.

edrumwri, 2017-02-24:
@sherm1, if we introduce position, velocity, and acceleration errors, then
we're back at the assumption that a system is mechanical.

--------------------------------
Evan Drumwright
Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2017-02-24:
Russ, there has been no mechanism planned for integrating systems with
constraints specified by inequalities. The upcoming piecewise DAE approach
will model inequality constraints in a Lagrange Multiplier type way (i.e.,
constraints can be active or inactive), which will admit this capability.

--------------------------------
Evan Drumwright
Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

On Fri, Feb 24, 2017 at 2:39 PM, Evan Drumwright <evan.drumwright@tri.global

RussTedrake, 2017-02-24:
@edrumwri - That's consistent with my understanding.  The first consumers of the inequality constraints will be the planning/optimization tools.

sherm1, 2017-02-25:
> if we introduce position, velocity, and acceleration errors, then
we're back at the assumption that a system is mechanical.

Good point. We had the same issue with continuous state variables but phrased them neutrally as second order state variables q, their related first order variables v, and other first order variables z. Is there analogous terminology we can use here for
- constraints on q, with t given: g₂(t;q)=0
    - holonomic, with time derivatives ġ₂(t,q;v), g̈₂(t,q,v;v̇)
- constraints on v, with t,q given: g₁(t,q;v)=0 
    - nonholonomic, with time derivative ġ₁(t,q,v;v̇)
- constraints on v̇,λ: g₀(t,q,v;v̇,λ)=0
    - e.g. sliding friction and softened g₂ and g₁ constraints

Also I am unclear whether we should allow constraints on z and if so how they should be specified.

RussTedrake, 2017-02-25:
my two cents:
- constraints on z should definitely be supported, too.  in matlab we had stateConstraints (phi(x)=0) for all Systems, and added additional notions of positionConstraints, velocityConstraints for Manipulators only.  [Adding a position constraint would also register a stateConstraint](https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/%40Manipulator/Manipulator.m#L163).  i think the semantics are slightly different here -- since any system can have q,v,z, but this is just to say that many non-manipulators announced constraints on z in systems that we care about.
- we already have a language for constraints in drake, and should definitely reuse that.  The existence (or non-existence) of derivatives should be expressed by the constraint class. 
- the constraint classes also allow constraints to be inequality constraints.  we should support that, too!  joint limits are a simple version.

RussTedrake, 2017-02-25:
in my mind, this is such an important thing to get right (to fulfill the promise of having a truly beautiful architecture for optimization over dynamical systems), that we should probably schedule a design meeting about it.  cc @hongkai-dai @soonho-tri

sherm1, 2017-02-25:
Good idea! @edrumwri and I may be out of action Monday due to HQ chaos on move-in day but should be functional again by Tuesday (I hope). We could also devote the Wed 10:30PT/1:30ET dynamics team meeting to this topic.

soonho-tri, 2017-02-25:
I'll read the discussions carefully.
/cc @jadecastro who is working on the design of `RequirementMonitor` with me.

> Good idea! @edrumwri and I may be out of action Monday due to HQ chaos on move-in day but should be functional again by Tuesday (I hope). We could also devote the Wed 10:30PT/1:30ET dynamics team meeting to this topic.

Tuesday is better for me and @jadecastro. On Wednesday, we have MIT project meetings (Ken and Larry visit us).

hongkai-dai, 2017-02-25:
I should be available on both Tuesday and Wednesday.

soonho-tri, 2017-03-15:
Can anyone summarize the action items (if any) that we agreed on in the last meeting?

sherm1, 2017-03-15:
> Can anyone summarize the action items (if any) that we agreed on in the last meeting?

I agreed to work on the specification of the equations that are represented by our System abstraction. I have a work-in-progress document [here](https://docs.google.com/a/trinst.com/document/d/1awnccf_3CVCIuiWBQNXs-ZlHPpmxkFSzaXstnRsGoLw/edit?usp=sharing) but it isn't ready yet. Feel free to add comments if you think of something.

The issue of how to handle input bounds constraints still remains. Russ & I discussed this morning. In the doc I'm assuming those would be a property of the input ports so that they would be easily visible. But that requires a flavor of CalcTimeDerivatives() (for example) that operates on the unbounded input values, so that an input-bounded linear system could reveal the linear part. The alternative is to depend on a System Diagram that has a subsystem that bounds its outputs that are fed into a linear subsystem. The difficulty there is attempting to discern that structure, while it would be obvious if the subsystem inputs declared it.

RussTedrake, 2017-04-30:
Besides the [document](https://docs.google.com/a/trinst.com/document/d/1awnccf_3CVCIuiWBQNXs-ZlHPpmxkFSzaXstnRsGoLw/edit?usp=sharing) @sherm1 referenced, #4998 and #4151 are also relevant.  I'm going to collect my thoughts in that document for now.

RussTedrake, 2017-06-24:
After some random conversations last week, and some upcoming use cases that we’d like to support, I propose that we move forward with the following initial API (which mirrors the new output port API) to System:

```
template <class MySystem, typename BasicVectorSubtype>
  const OutputPort<T>& DeclareConstraint(
      void (MySystem::*calc)(const Context<T>&, BasicVectorSubtype*) const,
      VectorXd lower_bound, VectorXd upper_bound) 
```

This does not satisfy all of the constraint types that @sherm1 articulated in the doc above, but it enables many immediately (input constraints, state constraints, parameter constraints).  I think that the block diagram logic will do a lot of the right work for us (for instance, if I use SymbolicExpression to ask a block diagram if it has any input port constraints, it will answer correctly).  

This should not be hard to do… and we can convert @edrumwri’s previous constraint implementations over to it.   Thoughts?

sherm1, 2017-06-26:
@RussTedrake, in your proposal above are you thinking of that as an easy-to-query *declaration*, independent of the internal constraint implementation?

sherm1, 2017-06-26:
A more-specific name like `DeclareBoxConstraint()` would be better, I think.

RussTedrake, 2017-06-26:
My goal was to constrain the function arguments (this captures all constraints that can be written as a function from Context to VectorXd.)  The internal implementation can be anything (not just a box constraint).  We can rely on symbolic expression to figure out the structure of the constraint.

My thinking is
(a) we want to support general constraints like this eventually.
(b) if we start breaking out box, linear, affine, quadratic, ... constraints, then we have to maintain them separately.  that can be a performance improvement after we get the initial version working.

sherm1, 2017-06-27:
(sorry for the slo-mo discussion rate) I don't think I understand what you mean by "constrain the function arguments" -- I don't see any enumerated arguments in the signature to constrain so I thought the bounds were on the BasicVectorSubtype output argument (which is why I thought it was a box). To what are the bounds applied? Can you clarify with some mathiness?

RussTedrake, 2017-06-28:
My goal was to describe a vector-valued constraint function of the (conceptual) form:
```
lower_bound <= f(const Context&) <= upper_bound
```
where `f()` is an arbitrary function of the context.  A bounding box constraint, a linear constraint, etc, would be special forms of this general case.

sherm1, 2017-06-28:
Oh, got it -- that makes sense. Bounds on the output of f. That's what I thought you meant but then I think I misused the term "box constraint" which I now think means that the bounds must apply directly to the decision variables. Is that right?

Above you said "my goal was to constrain the function arguments". I'm still confused by that statement. Did you mean "constrain the function result" or " constrain the argument which is a function"?

RussTedrake, 2017-06-28:
I could have chosen my words more carefully.  I just meant "let's implement 'f(const Context&)' first", even though it is probably not the best way to write friction cone or other acceleration-level constraints.

sherm1, 2017-06-28:
Thanks. I like the API. Am I correct in thinking of this as a declaration *about* the System's behavior rather than something that *causes* that behavior?

RussTedrake, 2017-06-29:
Good.  Yes, i think that it means "the system methods are only well-defined if this constraint is satisfied".   For instance, one should not pass in a non-unit quaternion, nor some other parameter out of range.  It also means, "you can safely analyze/optimize this system by only considering context values satisfying this constraint".

RussTedrake, 2017-08-26:
I’m chipping away at my proposed implementation.  Would be very interested in feedback if anybody is willing to take a quick look.  Some things will get better with additional syntactic sugar, but I’d like to get the basic organization right.  

https://github.com/RobotLocomotion/drake/compare/master...RussTedrake:system_constraints?expand=1

cc @sherm1, @jwnimmer-tri, @hongkai-dai, @soonho-tri

PR #12553 (2020-01-14): Unused constraint equations removal from system framework
Resolves #5297.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12553)
closed 2020-01-14 by jwnimmer-tri