RobotLocomotion/drake #5298: Injection of GLX_GLXEXT_LEGACY Preprocessor Flag Results in Conflict on Ubuntu 16.04

liangfok, 2017-02-24:
# Problem Definition

In #4930 we added the injection of the `GLX_GLXEXT_LEGACY` preprocessor flag to overcome the problem described in #4899. Unfortunately, I just discovered that the injection of this flag results in the following compile-time error indicating the flag was defined in other parts of VTK5:

```
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkColorMaterialHelper.cxx:22:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkFrameBufferObject.cxx:22:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkLightingHelper.cxx:18:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkLineIntegralConvolution2D.cxx:36:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkPixelBufferObject.cxx:22:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkSurfaceLICPainter.cxx:49:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkTextureObject.cxx:22:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkCameraPass.cxx:21:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkClearZPass.cxx:21:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkDepthPeelingPass.cxx:25:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkGaussianBlurPass.cxx:21:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkImageProcessingPass.cxx:21:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkLightsPass.cxx:21:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
[... 150 lines cut ...]
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Charts/vtkContextBufferId.cxx:21:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Charts/vtkOpenGLContextBufferId.cxx:23:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Charts/vtkOpenGL2ContextDevice2D.cxx:48:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/vtk/Charts/vtkOpenGLContextDevice2D.cxx:48:0:
/home/liang/dev/drake_catkin_workspace/build/drake/externals/vtk/Rendering/vtkgl.h:27:0: warning: "GLX_GLXEXT_LEGACY" redefined
 #define GLX_GLXEXT_LEGACY
 ^
```

To overcome this problem, I had to revert the injection of the `GLX_GLXEXT_LEGACY` flag (i.e., revert #4930) and manually uncomment line `drake_catkin_workspace/src/drake/externals/vtk/Rendering/vtkXOpenGLRenderWindow.cxx:30`, which defines `GLX_GLXEXT_LEGACY`.

I'm currently confused why this problem did not present itself back when #4930 was merged.

liangfok, 2017-02-24:
@maidenone, in
https://github.com/RobotLocomotion/drake/pull/4901#issuecomment-281494405, you mentioned that the injection of `GLX_GLXEXT_LEGACY` did not work. In what way did it not work? For example, did it result in the above "GLX_GLXEXT_LEGACY  redefined" error or did the processor simply think this macro wasn't defined?

maidenone, 2017-02-27:
I built with "cmake .. -DUSE_SYSTEM_VTK=OFF -DWITH_VTK=ON"

And i got something simular to this

In file included from /usr/include/GL/glx.h:333:0,
             from
/home/mildred/Work/3DKF/VTK/Rendering/vtkXOpenGLRenderWindow.cxx:31:
/usr/include/GL/glxext.h:480:143: error: ‘GLintptr’ has not been declared

I found the solution on this page:

http://stackoverflow.com/questions/28761702/getting-error-glintptr-has-not-been-declared-when-building-vtk-on-linux

I did only see this on 17.04, 16.10 worked fine.

liangfok, 2017-04-12:
I'm going to close this issue due to Director's imminent switch to VTK 6/7.

closed 2017-04-12 by liangfok