RobotLocomotion/drake #5307: Need Ability to Partially Load an SDF

liangfok, 2017-02-24:
# Problem Definition

Drake's SDF parser currently loads _every_ model defined in an SDF into the `RigidBodyTree` (soon to be `MultiBodyTree`). This is problematic when the SDF contains many models and certain systems within the simulation like controllers are only operating on one of the many models. 

# Proposed Solution

Update the SDF parser API to allow the user to specify which models within the SDF should be loaded into the `RigidBodyTree`.

sherm1, 2017-02-24:
I'm assuming it would be fine if `libsdformat` reads the whole sdf file as long as we can pull just the desired model(s) out of it. In that case from `libsdformat` we only need the ability to extract particular models from the in-memory representation. Is that feasible already, @clalancette?

liangfok, 2017-02-25:
Yes, it would be OK for `libsdformat` to load the entire SDF into memory, but then to selectively transfer only a portion of the models defined within the SDF into the `RigidBodyTree`.

liangfok, 2017-02-28:
This issue is blocked on https://github.com/RobotLocomotion/drake/issues/4337.

jwnimmer-tri, 2017-02-28:
On first read, within `parser_sdf.cc`, it would be trivial to add a guard to the call from `ParseSdf` to `ParseModel`, and a similar guard to `ParseWorld` if necessary.  Thus, I'm not sure why #4337 is a blocker.

liangfok, 2017-02-28:
Basically, I'm unsure whether it's worth touching `sdf_parser.cc` or `urdf_parser.cc` if they are going to be replaced by `libsdformat` in the near future.

jwnimmer-tri, 2017-02-28:
That depends on how much payoff vs work there is, but in any case is not "blocked" by 4337, since a fix is plausible even without #4337.

liangfok, 2017-02-28:
Perhaps I'll let @sherm1 and @siyuanfeng-tri collectively decide on the relative priority. It should be easy, but will require API changes since a new input parameter will need to be added to the parsers.

sherm1, 2017-02-28:
I'm not a consumer so have no opinion about the urgency of this feature. My preference would be to wait until we have switched to libsdformat if there is a reasonable workaround for now. Would splitting up the multi-model sdf into several smaller ones be a reasonable workaround?

jwnimmer-tri, 2017-02-28:
I would like to identify a first user who wants this feature, before we waste any more time discussing it.  So far, nobody on the thread fits that bill.

siyuanfeng-tri, 2017-02-28:
apologies for being late to the discussion. 
there is a desire from the manip group, but not a critical one. 
since this is mostly for facilitating working across sim and real hardware, and we have to setup an experimental repo for the hardware anyway, we can put xacro / giant sdf / duplicate models whatever there first.
my vote is we don't spend time on this now.

jwnimmer-tri, 2019-09-19:
Once someone needs this, it should be a relatively easy change to the `multibody::Parser` class.

closed 2019-09-19 by jwnimmer-tri