RobotLocomotion/drake #5317: Update SDF Parser To Allow Inclusion Of Other SDFs and URDFs. Then Disallow Programmatic Creation of RigidBodyTree.

liangfok, 2017-02-25:
# Problem Definition

We currently assemble the world by repeatedly adding model instances to a `RigidBodyTree` (soon to be `MultiBodyTree`), and instantiating things like sensors and actuators. While this works for simulations and controllers _within_ Drake, it does not work well with _external_ applications and tools like those in [OpenHumanoids](https://github.com/OpenHumanoids) and [Director](https://github.com/RobotLocomotion/director).

Drake has a custom URDF parser that is more expressive than SDF in many respects. We don't want to loose this expressiveness by switching entirely to SDF.

URDF is unable to express floating joints or multiple model instances without modifying the link names to include an "instance identifier". This prevents us from solely relying on URDF. SDF, on the other hand, does support the inclusion of multiple models within a single file.

# Proposed Solution

Let's update Drake's SDF parser to allow the inclusion of URDFs and other SDFs. This will enable us to describe the _entire simulated world_ using SDF and URDF files and thus initialize the `RigidBodyTree` via a **single call to the SDF parser**. We should thus **disable** the ability to load multiple model instances into the `RigidBodyTree` by calling the parsers multiple times. This programmatic method of loading model instances into `RigidBodyTree` is not sustainable in terms of allowing other tools like those in [OpenHumanoids](https://github.com/OpenHumanoids) and [Director](https://github.com/RobotLocomotion/director) to make use of the same model being used by Drake.

jwnimmer-tri, 2017-02-25:
I think there are valid use cases for defining multibody geometry on the fly, and just simulating / visualizing it.  We should not disallow that.

liangfok, 2017-02-25:
Would requiring all of those uses cases go through a serialization phase be too arduous? In other words, phase 1 is to generate the SDF + URDF files, then phase 2 is to simulate & visualize them.

jwnimmer-tri, 2017-02-25:
Yes, unacceptable.

liangfok, 2017-02-25:
Alright, I've updated the OP to not prevent users from initialziing a RigidBodyTree by repeatedly calling the parsers.

sherm1, 2017-02-25:
In what ways is URDF more expressive than SDF?

RussTedrake, 2017-02-25:
We added some drake-specific tags to urdf that we have not ported to sdf yet.  (springs, dampers, airfoils, cable-drives, etc).

liangfok, 2017-02-25:
I updated the OP with a link to Drake's URDF schema: http://drake.mit.edu/urdf/drakeURDF.html, which can be contrasted with ROS's URDF specification: http://wiki.ros.org/urdf/XML.

liangfok, 2017-02-26:
I was able to use SDF's `<include>...</include` tags to assemble a model containing two iiwa14s, two extra heavy duty tables, and one desk. The model is loadable in Gazebo, giving me some confidence that my SDF's syntax is correct:

[image]

One problem with my current SDF is it lumps all of the models into a single model in the world. I will next try to combine the `<world>...</world>` tags with the `<include>...</include>` tags to assemble a world consisting of multiple discrete models.

liangfok, 2017-02-27:
I figured out a working SDF syntax for including and instantiating separate SDF models. The example is in #5332. Now that I know what the `<include>...</include>` syntax looks like, we can begin modifying the SDF parser to support it.

One major question is whether we should wait for the new `libsdformat` library to be integrated (#4337) or modify Drake's current custom SDF parser.

liangfok, 2017-02-27:
Upon further reflection, I think it would be best to prioritize the switch to `libsdformat` prior to adding support for SDF files to include other SDF and URDF files. @sherm1 and @clalancette, please let me know if I should think otherwise. Thanks!

sherm1, 2017-02-27:
Sounds right to me, @liangfok. Any update @clalancette?

clalancette, 2017-03-01:
Hi @sherm1 ,
    Sorry I haven't had any updates for a while on this.  I've been stuck in sdformat land, trying to get rid of boost stuff.  I've made a lot of progress there, but haven't yet completed it.  Once that is done, I can get back to my PR to get sdformat integrated into drake.  I'll keep you informed.

amcastro-tri, 2018-07-11:
Closing this to track progress in this regard in the newer issue #8718 for MBP/sdformat.

closed 2018-07-11 by amcastro-tri