RobotLocomotion/drake #5325: use `AddConstraint(symbolic::Formula)` to handle all types of constraints

hongkai-dai, 2017-02-26:
<h2> Goal </h2>

Currently we support
```C++
AddLinearCost(symbolic::Expression);
AddQuadraticCost(symbolic::Expression);
AddLinearConstraint(symbolic::Formula); // linear equality constraint, linear inequality constraint, bounding box constraint
AddLorentzConeConstraint(symbolic::Expression e);
AddRotatedLorentzConeConstraint(symbolic::Expression e);
AddPositiveSemidefiniteConstraint(Eigen::Matrix<symbolic::Expression, Dynamic, Dynamic>);
```
We want to use `AddCost(symbolic::Expression)` and `AddConstraint(symbolic::Formula)` to handle all of these different forms, namely all costs and constraints in the supported convex optimization.

<h2> Approach </h2>

We will handle constraints and costs separately.

1. For cost, add a method
   ```C++
   AddCost(const symbolic::Expression& e)
   ```
   To handle both linear and quadratic cost, throw a runtime error if the input is not a linear or quadratic expression.  
**Question**: what is the return type for this function? I think `Binding<Constraint>` can work, but we lost information specific to the derived class like `LinearConstraint` and `QuadraticConstraint`.
2. For constraint, add a method
   ```C++
   AddConstraint(const symbolic::Formula& e)
   ```
   **Question**: What is a good way to write a conic constraint, such as Lorentz cone or semidefinite matrix cone? Shall we create a type of `symbolic::Formula` for `A*x+b is in K`, where `K` is a type of the cone?

hongkai-dai, 2017-02-26:
@soonho-tri and @RussTedrake for discussion.

RussTedrake, 2017-02-26:
I agree with all of the above.  My sense is that this is very similar to what is happening in `AddLinearConstraint` right now.  I agree that we will be throwing away return type information, but if the caller is being non-specific with the call, then presumably it will be ok being non-specific with the return type, too.

hongkai-dai, 2017-02-27:
Agree, I will first work on `AddCost(symbolic::Expression)` to handle both linear and quadratic costs.

PR #5340 (2017-03-02): Add symbolic cost
now `AddCost(const symbolic::Expression& e)` accepts both linear and quadratic cost. Partially resolves #5325

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5340)
closed 2017-03-02 by hongkai-dai