RobotLocomotion/drake #5327: Mysterious Build Error on OS X:  default initialization of an object of const type Foo without a user-provided default constructor

liangfok, 2017-02-26:
I am using 839b6e017bba05f92aaf6dc95f308c374f99d333 on OSX. 

# The Error

```
ERROR: /Users/liang/dev/drake-distro-3/drake/systems/framework/BUILD:491:1: C++ compilation of rule '//drake/systems/framework:value_checker_test' failed: sandbox-exec failed: error executing command /usr/bin/sandbox-exec -f /private/var/tmp/_bazel_liang/3302fbe3d4d2eae8487460393c40acaf/bazel-sandbox/8bbfb44b-02d0-4354-8ae2-f5e5a3578d43-7/sandbox.sb ... (remaining 70 argument(s) skipped).
drake/systems/framework/test/value_checker_test.cc:66:25: error: default initialization of an object of const type 'const drake::systems::(anonymous namespace)::NotSoGoodVector' without a user-provided default constructor
  const NotSoGoodVector not_so_good;
                        ^
                                   {}
drake/systems/framework/test/value_checker_test.cc:67:28: error: default initialization of an object of const type 'const drake::systems::(anonymous namespace)::NotQuiteGoodVector' without a user-provided default constructor
  const NotQuiteGoodVector not_quite_good;
                           ^
                                         {}
2 errors generated.
```

jwnimmer-tri, 2017-02-26:
The Continuous Production builds seem fine.  What is different between those builds and your OSX configuration?

liangfok, 2017-02-26:
Is there a `mac-bazel` in continuous production? I'm guessing #5303 is the difference. I just scheduled https://drake-jenkins.csail.mit.edu/view/Bazel/job/mac-clang-bazel-nightly-release/67 to see if it can replicate this issue.

jwnimmer-tri, 2017-02-26:
Perhaps OSX's clang is defective.  I guess the `= default;` for that constructor will need to change to `{}` to work around it.

liangfok, 2017-02-27:
The error was replicated on CI:

https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-clang-bazel-nightly-release/67/

I shall thus elevate this PR's priority.

PR #5328 (2017-02-27): Changes constructors of NotSoGoodVector and NoteQuiteGoodVector to be `{}` instead of `= default`.
Resolves #5327.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5328)
closed 2017-02-27 by liangfok