RobotLocomotion/drake #5330: Build failure

soonho-tri, 2017-02-27:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-clang-bazel-nightly-release/67

```
drake/systems/framework/test/value_checker_test.cc:66:25: error: default initialization of an object of const type 'const drake::systems::(anonymous namespace)::NotSoGoodVector' without a user-provided default constructor
  const NotSoGoodVector not_so_good;
                        ^
                                   {}
drake/systems/framework/test/value_checker_test.cc:67:28: error: default initialization of an object of const type 'const drake::systems::(anonymous namespace)::NotQuiteGoodVector' without a user-provided default constructor
  const NotQuiteGoodVector not_quite_good;
                           ^
                                         {}
2 errors generated.
```

This change was introduced by https://github.com/RobotLocomotion/drake/commit/f3476252560b1c9b3577ac15c674e6dc7db13d7a

/cc @kunimatsu-tri

liangfok, 2017-02-27:
I suspect this was already addressed by #5328.

liangfok, 2017-02-27:
I reopened this issue just in case.

soonho-tri, 2017-02-27:
@liangfok , thank you. I can confirm on my local mac that the problem is addressed by #5328. It's just the case that mac build takes hours in CI, and I got confused.

closed 2017-02-27 by soonho-tri