RobotLocomotion/drake #5341: Automotive demo cars rendering has wrong rotation axis

jwnimmer-tri, 2017-02-28:
(This is just from watching the demo, but I'm pretty sure I'm right.  Someone who does a code / model dive can confirm.)

The automotive demo's cars rendering uses the wrong yaw-rotation axis.  We are using the prius SDF model and mapping x,y,yaw to a floating joint.  A turning car should curve around the center of the rear axle, but to my eye we are pivoting around the rear bumper (approximately).

It's possible that this isn't worth fixing until PoseAggregator is in place, depending on how much rework is RigidBodyTree specific vs applicable to the PoseAggregator design as well.

/CC @david-german-tri @liangfok

liangfok, 2017-02-28:
I vaguely recall @maddog-tri saying something to this effect a while ago.

david-german-tri, 2017-02-28:
I don't understand why this would relate to PoseAggregator at all.

jwnimmer-tri, 2017-02-28:
The problem is not in the simulation, its only with what we send to DrakeVisualizer.  I assume that the refactoring of visualizer commands should happen first.

liangfok, 2017-03-08:
I suspect this may have to do with an incorrect assumption of the model wheelbase. The default wheelbase is `106.3 * 0.0254 = 2.70002 m`:

https://github.com/RobotLocomotion/drake/blob/b2b3fe8f469ab61840fe929035ef5348660f5ebd/drake/automotive/simple_car.cc#L293

Yet, the wheelbase of prius.sdf is only `0.17246968 + 1.40948 = 1.58194968 m`.

https://github.com/RobotLocomotion/drake/blob/0259ade9e281827d3b32d2c9b4c7e39e1e030c0f/drake/automotive/models/prius/prius.sdf#L47

https://github.com/RobotLocomotion/drake/blob/0259ade9e281827d3b32d2c9b4c7e39e1e030c0f/drake/automotive/models/prius/prius.sdf#L275

This discrepancy might explain why the pivot point is showing up behind the vehicle. If this is indeed the problem (still need to verify), we probably want to fix `prius.sdf` since a real Prius has a 2.7m wheelbase.

liangfok, 2017-03-09:
Looking into this more, I believe the problem is the root of the Prius model is in the center of the model, as shown by the following video:

https://youtu.be/iVKZXIhxBgQ

We need to change the root of the model to be in the middle of the rear axle. This would require resolving #2293. Alternatively, a quicker fix is to change the SDF model.

jwnimmer-tri, 2017-03-09:
A quicker fix is to do a frame shift between EFJS output and the D-V input.

liangfok, 2017-03-09:
I modified the Prius model to effectively position its root to be the middle of the rear axle:

https://youtu.be/RlaHXj1TeiE

I'll submit a PR with the fix to see if its something we want.

liangfok, 2017-03-29:
I am reopening this because the newly added `PriusVis` does not.contain the offset fix.

PR #5470 (2017-03-14): Adds ability to specify x-offset of vehicle visualization to ensure it matches the model's origin.
Resolves #5341. This is an alternative to #5457.

Here's what `SimpleCar` looks like with the correct x-offset applied:

https://youtu.be/iDpAANTiuPU

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5470)
PR #5658 (2017-03-30): Adds offset between the vehicle model's frame and the Prius visualization's frame.
Resolves #5341. Hopefully for real this time.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5658)
closed 2017-03-30 by liangfok