RobotLocomotion/drake #5343: Provide System-level mechanism for Allocating inputs

RussTedrake, 2017-02-28:
For motivation, see:
https://github.com/RobotLocomotion/drake/blob/master/drake/systems/trajectory_optimization/direct_collocation.cc#L28

This toolchain needs to be able to create inputs to supply to the provided System*, and currently simply creates BasicVector's because it doesn't know any better.  This is insufficient for systems that expect typed input, e.g.,
https://github.com/RobotLocomotion/drake/blob/master/drake/automotive/simple_car.cc#L109

RussTedrake, 2017-02-28:
I would like this to live at the `System` level.  I feel that there are two natural choices:
1) define   
```
virtual std::unique_ptr<SystemOutput<T>> AllocateInput(
      const Context<T>& context) const = 0;
```
and  answer the question about whether we re-use the `SystemOutput` type or actually create a new `SystemInput` type (that is essentially identical).  Possibly `typedef SystemOutput SystemInput` could make the code more palatable. 

2) provide 
```
  virtual std::unique_ptr<BasicVector<T>> AllocateInputVector(
      const InputPortDescriptor<T>& descriptor) const {
    return std::make_unique<BasicVector<T>>(descriptor.size());
  }

  virtual std::unique_ptr<AbstractValue> AllocateInputAbstract(
      const InputPortDescriptor<T>& descriptor) const {
    DRAKE_ABORT_MSG("A concrete leaf system with abstract input ports must "
                    "override AllocateInputAbstract.");
  }
```
at the `System` level.  The only thing that I don't like about that is the lack of parallelism with the current `System`/`LeafSystem` design.  `AllocateOutputVector` is currently only defined for `LeafSystem`.  But apart from the lack of parallelism, I actually think it's the right thing to do, so will PR that first.
@david-german-tri , @sherm1 -- would appreciate your comments.

RussTedrake, 2017-02-28:
WIP for consideration:
https://github.com/RobotLocomotion/drake/compare/master...RussTedrake:allocate_input?expand=1
If people approve of this design, then we can add the unit tests.

jwnimmer-tri, 2017-02-28:
Approach (2) seems like the right System API to me.

The only other big question to me is whether `CreateDefaultContext` should end up with freestanding correctly-typed input ports by default, or nullptr inputs like it does now.  I am hard-pressed to come up with reasons why anyone who asks a System to make a Context would want null inputs in general; it seems much more reasonable to provide default freestanding inputs wired up, and then the caller can `set_value` them if they want something else (as with any other state, or params, etc).

The exception to that rule would be for Systems with _optional_ input ports, where the behavior of the system changes when a port is disconnected versus all-zeros.  In that case, perhaps `AllocateVectorInput` should be returning nullptr, though that doesn't meet the need for abstract-valued optional inputs.  Perhaps we should dismiss inferred-by-null optional inputs anyway, and require System authors to explicitly ignore inputs rather than try-null use them.

RussTedrake, 2017-02-28:
I think the reason to have nullptr inputs is because of the (most common) input case where we are wiring the output of another system into the input of the current system.  Why incur the cost of allocating an input vector if it's about to be thrown away?

jwnimmer-tri, 2017-02-28:
Perhaps.  But `CreateDefaultContext` should be a rare call -- `Context::Clone` is what simulation or control loops should be using.

jwnimmer-tri, 2017-02-28:
Maybe the compromise is to have `System::CreateDefaultContext` leave the inputs null, but offer a `System::AllocateInput(Context*)` that allocates them into a passed-on context, using the `System`'s allocators with freestanding input ports.

david-german-tri, 2017-02-28:
> Maybe the compromise is to have System::CreateDefaultContext leave the inputs null, but offer a System::AllocateFreestandingInputs(Context*) that allocates them into a passed-in context, using the System's allocators with freestanding input ports.

I like this, because I don't want to give up on optional input ports, because it separates shape from content, and because System authors that don't have a need to implement it experience no additional overhead.

david-german-tri, 2017-02-28:
Also, in the motivating dircol example, why not just specify that appropriate inputs are already hooked up to the Context, and use `system_->EvalVectorInput`?

sherm1, 2017-03-01:
Sorry I'm late to this party. Thoughts:
- David's pending question makes sense to me -- EvalVectorInput() already returns a reference to a value of the right type assuming all the wiring is complete. Is that enough?
- Otherwise I don't think it is unreasonable to ask an InputPort to create a value it would find acceptable. But that seems somewhat unsatisfying since there could be many different acceptable inputs and an input port can reasonably accept any output if all it does is pass it along or sample it.

jwnimmer-tri, 2017-03-01:
In other words, the counter-proposal for the motivating example is that the Context that is passed must be wired up to inputs somehow, and then internally we Eval and then Clone those input values to become freestanding ports in our internal Context?

sherm1, 2017-03-01:
Right, just as the motivating example is doing for time, state, and parameters:
```c++
context_->SetTimeStateAndParametersFrom(context);
```
could be
```c++
context_->SetTimeStateParametersAndInputsFrom(context);
```

PR #5357 (2017-03-04): Allocate input
As discussed with @jwnimmer-tri and @david-german-tri.  Resolves #5343

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5357)
commit 89f81ce1 (2017-03-04): Allocate input (#5357)

Adds a mechanism for a system to allocate a specialized input type.  Resolves #5343.
closed 2017-03-04 by david-german-tri