RobotLocomotion/drake #5344: LeafSystem::AllocateOutput should be final

RussTedrake, 2017-02-28:
Digging around to decide on the implementation details for #5343, I was troubled to see the existence of both `AllocateOutput` and `AllocateOutputVector` for `LeafSystem`.  Based on my quick inspection, it seems that if a derived system overloads only `AllocateOutput` because it has a special type, then calls to `AllocateOutputVector` could do the wrong thing, and I don't see any guarantee that `AllocateOutputVector` won't be called (it is `protected`).  

I *think* that `AllocateOutput` should be declared final in `LeafSystem`, enforcing that derived classes only overload `AllocateOutputVector`.

jwnimmer-tri, 2017-02-28:
Since both `AllocateOutputAbstract` and `AllocateOutputVector` exist, subclasses should be able to do all they need by overriding one of both of those, so yes I agree that we should make `AllocateOutput` final.

david-german-tri, 2017-02-28:
Yeah, I agree; this is an artifact from before `AllocateOutput{Vector,Abstract}` existed.

PR #5348 (2017-03-02): Make LeafSystem::AllocateOutput final.
Remove some legacy, boilerplate AllocateOutput overrides.

@amcastro-tri and @liangfok for feature review, since the only nontrivial change diffs are in RigidBodyPlant and LcmSubscriberSystem.  They are intended to be no-ops.

Fixes #5344.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5348)
closed 2017-03-02 by david-german-tri