RobotLocomotion/drake #5345: Add operator== and operator!= between symbolic expression matrices

soonho-tri, 2017-02-28:
Feature requested by @RussTedrake.

/cc @hongkai-dai

soonho-tri, 2017-02-28:
FYI, we already have `operator{==,!=}`  between `Eigen::Matrix<symbolic::Expression>`s. They are provided by `Eigen::MatrixBase<T>::operator{==,!=}`. Note that these operators return `bool` and check *structural* equality/inequality by relying on `Expression::EqualTo`.

Russ's following example failed because we don't have `AddLinearConstraint(bool)`.

```c++
AddLinearConstraint(direct_traj.x(0) == Vector2<symbolic::Expression>::Zero());
```

I think we need one (or both) of the followings:

1. `eq₁`, formula-forming comparison returning `Formula`.
    For example, `eq₁([x y], [1 2])` returns a formula `(x == 1) && (y == 2)`.

2. `eq₂`, element-wise comparison returning `Eigen::Matrix<Formula>`. 
    For example, `eq₂([x y], [1 2])` results in a matrix `[(x == 1)   (y == 2)]`.

Questions
----------

1. Is there a preference between `eq₁` and `eq₂`?
2. Do we need to re-assign a function (eq₁ or eq₂) to `operator==`? I still need to check the consequence of this possible change.

soonho-tri, 2017-02-28:
@hongkai-dai , we can add similar functions for `operator<=`. FYI, Eigen does not provide operator overloading for other relational operators (`<`, `<=`, `>`, `>=`).

hongkai-dai, 2017-02-28:
@soonho, I prefer `eq2`. If we use `eq1`, then `AddLinearConstraint(Formula)` can also accept formula like `(x == 1) || (y == 2)`, and we can only throw a runtime error  for this case, but we cannot detect this at compile time. Using `eq2`, we know that we only accept that ALL comparisons are `==`, `<=` or `>=`.

RussTedrake, 2017-02-28:
I'm not sure that i've thought through all of the consequences, but i agree ... eq2 is what i had imagined would be the right approach

soonho-tri, 2017-03-08:
@RussTedrake , could you take a look at the PR #5445?
In summary, we have:
 - `eq₁` becomes  `operator==` between `Eigen::Matrix<symbolic::Expression>`.
 - `eq₂ ` becomes `operator==` between `Eigen::Array<symbolic::Expression>`.

soonho-tri, 2017-03-09:
Closed by #5445.

closed 2017-03-09 by soonho-tri