RobotLocomotion/drake #5359: Bazel creates lcm python bindings in the wrong scope

sammy-tri, 2017-03-01:
LCM Python code expects to say things like
```import drake.lcmt_polynomial```
and get a class, not a module.  Without generating the appropriate ```__init__.py``` as the CMake builds do, that import gets the module, and the class winds up as drake.lcmt_polynomial.lcmt_polynomial

Will attach a patch to reproduce.  Note that this breaks the generated encode() message too (among other things).
[broken_lcm_exports.diff.txt](https://github.com/RobotLocomotion/drake/files/811183/broken_lcm_exports.diff.txt)

Reproduce with ```bazel run //drake/lcmtypes:broken_emitted_python```

jwnimmer-tri, 2017-03-03:
I've pushed the test case to https://github.com/jwnimmer-tri/drake/tree/bazel-lcmtypes-module and confirmed I can repro it.  I will investigate.

jwnimmer-tri, 2017-03-03:
So the situation is that, if you have say these LCM messages defined:
- package drake; struct lcmt_foo ...
- package drake; struct lcmt_bar ...
- package drake; struct lcmt_baz ...

Then the import statements that LCM expects / provides to use them is ...
- import drake.lcmt_foo
- import drake.lcmt_bar
- import drake.lcmt_baz

Then the filesystem situation is ...
- drake/\_\_init__.py
- drake/lcmt_foo.py
- drake/lcmt_bar.py
- drake/lcmt_baz.py

And the contents of drake/\_\_init__.py are:
```python
"""LCM package __init__.py file
This file automatically generated by lcm_wrap_types.
DO NOT MODIFY BY HAND!!!!
"""

from .lcmt_foo import lcmt_foo
from .lcmt_bar import lcmt_bar
from .lcmt_baz import lcmt_baz
```

The Bazel rules aren't generating the init.py file right now.  Moreover, generating that file implies that all lcmtypes in package `drake` are a single rule, which is frustrating.  Namespace and libraries should not be identical concepts!  Especially when namespaces don't nest!

I am considering how to best fix this.

jwnimmer-tri, 2017-03-04:
So actually this is not so bad.

Problem (1) was that `lcm-gen` was producing the `__init__.py`, but I didn't _tell_ Bazel that was doing so as part of the action, so Bazel effectively deleted it because it's a pedant (yay!).  Oops.

Onces (1) is fixed, problem (2) is that now all of the _multiple_ `lcm_py_library` calls we have all want to own the creation `drake/__init__.py` and try to stop on each other.  So Bazel's dependency analysis says "Yo dude, these rules have a feud" and I'm like "Right!  Rock on, that's exactly the problem at hand". So the solution is to only have _one_ `lcm_py_library` call per lcm_package per subdirectory.

PR forthcoming.

PR #5403 (2017-03-06): Put all lcmtypes-py into a single library
This is because the `drake/__init__.py` is generated with a shim for each struct, so we need to do the whole lcm_package all at once.

Fixes #5359.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5403)
closed 2017-03-06 by jwnimmer-tri