RobotLocomotion/drake #5390: Need a maliput::api::Lane API That Provides Motion Derivative in Geo-Space

liangfok, 2017-03-03:
# Problem Definition

Maliput lanes currently provide a way to get motion derivates via [maliput::api::Lane:::EvalMotionDerivatives()](http://drake.mit.edu/doxygen_cxx/classdrake_1_1maliput_1_1api_1_1_lane.html#ab82fa390b6bc325473c13dc85dd02401). This method, however, returns the motion derivatives in lane-space `(s, r, h)`. To integrate with car models that operate in geo-space `(x, y, z)`, we need a version of this method that returns the motion derivatives in geo-space.

# Suggested Course of Action

Add a method called `maliput::api::Lane::EvalGeoMotionDerivatives()` that takes the same parameters as [maliput::api::Lane:::EvalMotionDerivatives()](http://drake.mit.edu/doxygen_cxx/classdrake_1_1maliput_1_1api_1_1_lane.html#ab82fa390b6bc325473c13dc85dd02401) but returns the motion derivatives in geo-space.

jwnimmer-tri, 2017-03-11:
@liangfok Is this issue still accurate?

liangfok, 2017-03-11:
For `MaliputRailcar`, we decided to keep the vehicle model's state in lane-space (i.e., using `s` and `s_dot`) and thus can make do with the existing [maliput::api::Lane:::EvalMotionDerivatives()](http://drake.mit.edu/doxygen_cxx/classdrake_1_1maliput_1_1api_1_1_lane.html#ab82fa390b6bc325473c13dc85dd02401). I shall thus close this issue.

closed 2017-03-11 by liangfok