RobotLocomotion/drake #5394: How to add moment disturbance

GuiyangXin, 2017-03-03:
Hi all,

I know we use Copy_of_three_dof_force.urdf to add external force to Atlas or Valkyrie. And I have implement it on LittleDog successfully according to pushing recovery example of Atlas. Now I want to add moment disturbance to LittleDog. How can I do? How to modify this file Copy_of_three_dof_force.urdf? Thanks.

hongkai-dai, 2017-03-11:
Sorry for my belated reply. I think if you have two force element at two different positions, with opposite direction and the same magnitude of the force, you will then get a torque.

So maybe you can modify the three_dof_force.urdf as
```
<force_element name="fx_plus">
    <thrust>
      <parent link="three_dof_force_link"/>
      <origin xyz="0 0 0.1"/>
      <axis xyz="1 0 0"/>
    </thrust>
  </force_element>

<force_element name="fx_minus">
    <thrust>
      <parent link="three_dof_force_link"/>
      <origin xyz="0 0 -0.1"/>
      <axis xyz="-1 0 0"/>
    </thrust>
  </force_element>
```
This will create a torque in the `y` direction, if the forces have the same magnitude, but opposite direction.

GuiyangXin, 2017-03-11:
Yes, a smart way. I will have a try. Thank you very much, Hongkai.

closed 2017-03-11 by hongkai-dai