RobotLocomotion/drake #5406: Remove libdreal build system integration

jwnimmer-tri, 2017-03-04:
Drake's current dReal integration uses CMake to bring in https://github.com/dreal/dreal3 and run a single unit test; `libdreal` is not yet bridged into the MathematicProgram APIs (it just throws an exception).  Our roadmap has us using Bazel to bring in dReal (instead of CMake) and https://github.com/dreal/dreal3-apache2 (instead of dreal3 GPL).

I propose that the current dReal integration in Drake isn't worth the trouble to keep working during the transition to Bazel / Apache2, and that we should disable it to hasten the Bazel transition.

Specifically, I propose that we:
- Remove the code for the superbuild and drake-ci implementations of `WITH_DREAL`.
- Remove the git submodule `drake-distro/externals/dreal`.
- Remove `drake/solvers/test/dreal_solver_test.cc` from the tree (because is needs `libdreal`)
- Remove `drake/solvers/dreal_solver.cc` from the tree (because it wouldn't be covered by CI).
- Do _not_ remove the `no_dreal.cc` stub nor the `kDreal` labels in the solver, etc.  They are reasonably likely to be used in the future, will be covered by CI (I think?), and shouldn't get in our way.

@david-german-tri @soonho-tri What do you think?

soonho-tri, 2017-03-04:
SGTM. I'll take care of the items above.

david-german-tri, 2017-03-06:
SGTM as well.

closed 2017-03-22 by soonho-tri