RobotLocomotion/drake #5408: Keep Acrobot model files in sync

jwnimmer-tri, 2017-03-04:
In https://github.com/RobotLocomotion/drake/tree/master/drake/examples/Acrobot, we have `Acrobot.sdf` and `Acrobot.urdf`.  The files should be interchangeable, so we should ensure that the two files remain sync'd, in terms of the resulting MultibodyTree and their dynamics.

We had a program that performed this check at https://github.com/RobotLocomotion/drake/blob/e8ecbe96a6608b0ab61390b3dce06b6dc3fe4bfe/drake/multibody/rigid_body_system1/test/compareRigidBodySystems.cpp, but it got overtaken by events.  We should consider reimplementing it, or else removing one of the two duplicated model files, or etc.

amcastro-tri, 2017-03-06:
The [acrobot benchmark](https://github.com/RobotLocomotion/drake/tree/master/drake/multibody/benchmarks/acrobot) will probably prove useful in verifying the two different models.

amcastro-tri, 2017-03-06:
cc'ing @clalancette

EricCousineau-TRI, 2019-03-25:
Per #11009 / #11011, I want to delete `examples/Acrobot.{urdf,sdf}` and only use the benchmark versions.

Once the `examples` models are deleted / fully slated for deprecation (and not used at all), we can close this issue.

jwnimmer-tri, 2019-03-25:
@RussTedrake cares a lot about what the examples look like.  I suggest getting his explicit approval on any plan that touches exapmles/acrobot, before merging any PRs.

RussTedrake, 2019-03-26:
Just answered in #10294 .  There is downstream code that depends on these (in underactuated).

jwnimmer-tri, 2019-04-23:
Due to #6619 and #10294, I should probably own this issue as well.

jwnimmer-tri, 2020-09-28:
Re-assigning to @mitiguy to resolve.  In #13593 we're apparently adding yet another acrobot, buried inline in a unit test.  That's beyond what I can retrofit.

mitiguy, 2020-09-29:
@jwnimmer-tri   I am _not_ creating another acrobot in PR #13593.   I think it important to be able to create small hand-crafted fixtures/models (e.g., a kinematic-only double pendulum) for testing purposes without reliance on other models that have other dependencies.

To make this point clearer, I am adding the following comment to the test fixture (model):
// Note: The applied forces (such as gravity) on this system are irrelevant as
// the plan for this text fixture is limited to testing kinematics for an 
// instantaneous given state.  The mass/inertia properties of this plant are
// irrelevant and there are no actuators, sensors, controller, or ports.

Note: As you deem best, feel free to unassign this issue to me.

jwnimmer-tri, 2020-09-29:
I'll just close this as a duplicate of #10294.  I'm not sure that the SDF vs URDF sync is the big deal, vs the C++ implementations.  If we do care about the files in particular, we can have it as a subtask of #10294.

closed 2020-09-29 by jwnimmer-tri