RobotLocomotion/drake #5412: dependency install for under actuated robotics class in Edx

junhyeokahn, 2017-03-05:
Hi,
First of all thank you for your efforts to maintain this awesome hands on codes.

I am using  ```91d5a18``` version for underactuated  robotics class in edx.
And I already installed mosek and gurobi and placed license files on ```/$HOME/gurobi.lic``` and ```/$HOME/mosek/mosek.lic```
Also in CmakeLists.txt, I add 'ON' to the option(with_gurobi, and with_mosek)

```
cmake_minimum_required(VERSION 3.5)
project(drake-superbuild)

include(CTest)

include(cmake/config.cmake)
include(cmake/externals.cmake)
include(cmake/examples.cmake)
include(cmake/git/hooks.cmake)

drake_setup_superbuild()
drake_setup_platform()
drake_setup_git_hooks()

###############################################################################
# BEGIN options

##########################################
# External Projects that are ON by default
##########################################
option(WITH_EIGEN "required c++ matrix library.  only disable if you have it already." ON)
option(WITH_GOOGLETEST "required c++ unit test library.  only disable if you have it already." ON)
option(WITH_GFLAGS "required c++ command-line library.  only disable if you have it already." ON)
option(WITH_GOOGLE_STYLEGUIDE "provides cpplint.py style checking" ON)
option(WITH_SWIGMAKE "helper tools to build python & MATLAB wrappers for C++ libraries with Eigen" ON)
option(WITH_BULLET "used for collision detection" ON)
option(WITH_LCM "interprocess communications protocol for visualizers, etc" ON)
option(WITH_BOT_CORE_LCMTYPES "required LCM types library. only disable if you have it already." ON)
option(WITH_SPDLOG "spdlog text logging facility; disabling will turn off text logging." ON)

if(WIN32)
  option(WITH_GTK "precompiled gtk binaries/headers for Windows" ON)  # needed for lcm on windows
else()
  option(WITH_DIRECTOR "vtk-based visualization tool and robot user interface" ON) # not win32 yet.  it builds on windows, but requires manually installation of vtk, etc.  perhaps make a precompiled director pod (a bit like snopt)
  option(WITH_LIBBOT "simple open-gl visualizer + lcmgl for director" ON) # there is hope, but i spent a long time cleaning up c/c++ language errors trying to make msvc happy.. and still had a long way to go.
  option(WITH_NLOPT "nonlinear optimization solver" ON)
  option(WITH_DREAL "nonlinear SMT solver" OFF)
  option(WITH_MOSEK "convex optimization solver; free for academics" ON)
  # IPOPT is currently disabled on Mac due to MATLAB compatibility
  # issues: https://github.com/RobotLocomotion/drake/issues/2578
  if (NOT APPLE)
    option(WITH_IPOPT "nonlinear optimization solver" ON)
  endif()
  option(WITH_SWIG_MATLAB "A version of SWIG with MATLAB support" ON)
endif()

option(WITH_SNOPT_PRECOMPILED "precompiled binaries only for snopt; the source requires a license (will be disabled if WITH_SNOPT=ON)" ON)
option(WITH_YAML_CPP "library for reading and writing yaml configuration files" ON)

##############################################################
# External Projects that are only needed when MATLAB is in use
##############################################################
include(CMakeDependentOption)
option(DISABLE_MATLAB "Don't use MATLAB even if it is present." OFF)
if(DISABLE_MATLAB)
  message(STATUS "MATLAB is disabled.")
else()
  find_program(matlab matlab)
  if(matlab)
    message(STATUS "Found MATLAB at " ${matlab})
  else()
    message(STATUS "Looked for MATLAB but could not find it.")
[... 316 lines cut ...]

## grab and install precompiled snopt

# TODO: look for snopt_c
if(snopt_c_FOUND OR WITH_SNOPT)
  set(WITH_SNOPT_PRECOMPILED OFF)
endif()
if(WITH_SNOPT_PRECOMPILED)
  message(STATUS "Preparing to install precompiled snopt")
  ExternalProject_Add(download-snopt-precompiled
    URL "https://s3.amazonaws.com/drake-provisioning/drakeSnopt.zip"
    URL_MD5 "7b36168cba2fb9a56b2fd6117427fc4a"
    SOURCE_DIR "${CMAKE_BINARY_DIR}/snopt-precompiled"
    CONFIGURE_COMMAND ""
    BUILD_COMMAND cmake -E copy_directory
      ${CMAKE_BINARY_DIR}/snopt-precompiled/
      ${PROJECT_SOURCE_DIR}/drake/matlab/solvers/
    INSTALL_COMMAND "")
  add_dependencies(download-all download-snopt-precompiled)
  add_dependencies(drake download-snopt-precompiled) # just in case: make sure any compiled drake version happens after precompiled install
endif()
```
But when I try to run commonLyapunovForLinearSystems or Atlas example, it returns I don't have dependency. Also when I explore external/mosek or external/gurobi directory there is nothing there.

I would appreciate if you give me a help.

hongkai-dai, 2017-03-05:
`mosek` and `gurobi` are proprietary softwares. They can be obtained in https://mosek.com/ and https://www.gurobi.com/ respectively.

junhyeokahn, 2017-03-05:
Thanks for reply!

Yes I installed them both from those site.
How do I change CMakeLists.txt after that to use those?

hongkai-dai, 2017-03-05:
You will need to make sure that the mosek file `mosekopt.m` and `gurobi.m` are in your MATLAB path, you may want to check `checkDependency.m`, line 214 and line 239 for the detail.

junhyeokahn, 2017-03-05:
I guess I wrongly download mosek and gurobi.
```mosekopt.m``` and ```gurobi.m``` is comming when I download from the website?
I actually couldn't see those files

hongkai-dai, 2017-03-05:
For mosek, when I download it, I can find mosek.m in `mosek/7/toolbox/r2014a`. The last `r2014a` is the MATLAB version I choose.

For gurobi, I can see it in `gurobi/gurobi605/linux64/matlab/gurobi.m`.

junhyeokahn, 2017-03-05:
Thank you so much,
I got it :)

closed 2017-03-05 by hongkai-dai