RobotLocomotion/drake #5431: LeafSystem should support model-value based input/output port allocation

jwnimmer-tri, 2017-03-07:
LeafSystem should support model-value based input/output port allocation; it already supports this for the continuous and discrete state.

See also https://en.wikipedia.org/wiki/Prototype_pattern.

jwnimmer-tri, 2017-03-07:
PR train WIP at https://github.com/jwnimmer-tri/drake/tree/framework-model-io-ports

sherm1, 2017-03-07:
@jwnimmer-tri, I am enthusiastic about model-based specification for port values. Looking at #5432 (and possibly misunderstanding it) it appears that you are preparing to provide an array of model values rather than associating a single model value with each port. Is that right?

jwnimmer-tri, 2017-03-07:
Nope; each port will have a single model value, to be provided at declaration-time if at all.  You can browse the commits in the WIP link; it's spike-tested all the way through to SimpleCar as an example.

jwnimmer-tri, 2017-03-08:
> I love the idea of using model values to specify allocation type for output ports.
>
> For input ports we need a definition of acceptable types; a model can be used as a very specific requirement but it's not good for an input port that can accept a variety of sources (e.g. pass-through or sampling Systems). But a model could be used to provide default values for input ports, in which case the model is really specifying "output" values that the input ports are to receive.

I think this is all correct, and the second part is what's going on here.  In particular, we need to customize [System::AllocateInputVector](http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_system.html#af2afe75cbb120c69321970a26c62c4c8)'s result when `LeafSystems` need subclasses of `BasicVector` to appear at their inputs.  Right now, that's a gross override; I think its much better as a model value.  If we need to change the input port semantics to inherit types or sizes or whatever based on their connection, then `AllocateInputVector` will have to change, and this model-based input idiom will follow suit.

> One thing I don't love is treating the input ports as a collection and the output ports as a collection. Each port is independent and it is awkward to treat them together since (for example) it obfuscates their dependencies.

That's a fair point, but `System` (and thus `LeafSystem`) use arrays now, so the near-term resolution this issue will keep that design intact.  If in some far-future world the System vs Port design evolves, the model-based concept should track it nicely -- and probably even makes it easier, because `LeafSystem` will have less brittle and redundant ceremony around their port types, so less boilerplate to edit as APIs change.

PR #5462 (2017-03-14): Add model values for input and output ports
Fixes #5431.

Add model values for input and output ports.
Update simple_car to use model ports.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5462)
closed 2017-03-14 by jwnimmer-tri