RobotLocomotion/drake #5469: Value<T> should support T's that implement Clone

jwnimmer-tri, 2017-03-10:
Right now, `Value<T>` only supports `T`'s that are copyable (using C++ copy constructor / copy-assignment operator).  It would be very nice for users if `T`'s that quack like a `std::unique_ptr<T> Clone() const` member method could be used as well, and especially that the `AbstractValue` methods like `template <class T> const T& AbstractValue::GetValue()` also worked seamlessly (with exposing some kind of `unique_ptr` intermediary).  I think should should all be possible with a bit of SFINAE and type_traits.

At the same time, we might also consider if we could offer `AbstractValue` some kind of move semantics, e.g. `AbstractValue::MoveFrom` and friends to mimic `AbstractValue::SetFrom`.  Or at least, consider that future possibility when updating the code for `Clone`-able `T`'s.

SeanCurtis-TRI, 2017-04-04:
This seems to solve the following problem:

Given the class:

```
class Foo {

   unique_ptr<Bar> my_bar_;
};
```

I'd like to be able to have a `Value<Foo>`, but I can't (as presented).  It has no assignment operator or copy constructor. There are three possible solutions.

1) Simply implement `Foo(const Foo& f) const` and `Foo& operator=(const Foo& f)` explicitly.
2) Implement a `unique_ptr<Foo> Clone() const` method on `Foo`, and make `Value<T>` smart enough to use it in its own assignment operator/copy constructor.
3) Introduce `copyable_unique_ptr<T>` as a `unique_ptr` with a default copy constructor and assignment operator.  It would be smart enough to find either a copy constructor for `T` or a `Clone()` method on T (the same smarts as in #2). Change `unique_ptr<Bar>` to `copyable_unique_ptr<Bar>` in the definition of `Foo`.

Implications of the options:
1. Requires implementing copy constructor and assignment operator for a class that, but for a `unique_ptr`, would happily have a default constructor. The two methods are somewhat redundant (although this can be somewhat mitigated).
2. Requires SFINAE smarts to prefer `Clone()` over copy constructor, but use either one if only one is available. Still requires the template parameter `T` to have one of those methods. So, it would still require `Foo` to explicitly implement `Clone()` or copy constructor.
3. Requires the same SFINAE smarts as 2. However, `Foo` would not have to explicitly define any method (`Clone`, assignment, or copy constructor) because the defaults would be valid.

Certainly, making `Value<T>` smarter is good -- passing in sub-classes as values that can only be correctly copied via calls to `Clone()` would definitely benefit from that. But it only *indirectly* address the problem with class `Foo`. It still requires introducing non-default functionality in `Foo`. The `copyable_unique_ptr` elegantly handles that.

Given that we can steal wholesale from simbody, it's a low cost to introduce as well (https://simbody.github.io/simbody-master-doxygen/api/classSimTK_1_1ClonePtr.html#details).

At the same time, we can also introduce the copy-on-write `unique_ptr` (https://simbody.github.io/simbody-master-doxygen/api/classSimTK_1_1CloneOnWritePtr.html#details) which even comes with unit tests (https://github.com/simbody/simbody/blob/master/SimTKcommon/tests/TestCloneOnWritePtr.cpp).

SeanCurtis-TRI, 2017-04-04:
/cc @sherm1 FYI

jwnimmer-tri, 2017-04-04:
The purpose of this issue is that sometimes you have a `T` that is `Clone`-able but should not be copyable, and we want to have `Value<T>` be convenient without users having to manually wrap `T` to be copyable -- the framework should perform that adaptation internally.

That is a different problem than imbuing actual copy and copy-assign semantics onto types that have `unique_ptr<Something>` as a member field.

sherm1, 2017-04-04:
FTR, I like the idea of making `Value<T>` smarter as proposed here but would prefer not to mix that discussion with `copyable_unique_ptr<T>` which should live or die on its own merits.

SeanCurtis-TRI, 2017-04-05:
@jwnimmer-tri Out of curiosity, if you have a class that is both (copy constructor and clone), would you favor the copy-constructor over the clone?

jwnimmer-tri, 2017-04-05:
I think copy, yes.  The main distinction would be whether the `Value<>` has `T` or `unique_ptr<T>` as the member field, and I guess the direct member seems like the better choice.

SeanCurtis-TRI, 2017-04-05:
Good.  That's what I was thinking as well.

jwnimmer-tri, 2017-05-17:
WIP at https://github.com/jwnimmer-tri/drake/tree/framework-value-natively-cloneable

I think it is correct, but could perhaps be a bit more optimized to reduce the amount of cloning, and I don't have anywhere near enough tests yet.

/CC @siyuanfeng-tri

jwnimmer-tri, 2017-05-18:
I think this is done on the WIP branch now, just need to push it through PRs.  It isn't optimized, and doesn't have sugar for extra downcasting, but should be a solid starting point.  Once we have some more use cases ready, we can look at sugar for polymorphic clone-only types.

PR #6177 (2017-05-26): Add Value<T> support for clone-only types
Resolves #5469.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6177)
closed 2017-05-26 by jwnimmer-tri