RobotLocomotion/drake #5487: nonlinear_program_test failure under Ubuntu 16.04 with g++-4.9

jwnimmer-tri, 2017-03-14:
When I run `bazel test -c dbg --compiler=gcc-4.9 --config memcheck //drake/solvers:nonlinear_program_test` on Ubuntu 16.04 Xenial, I get:
```
==7== Conditional jump or move depends on uninitialised value(s)
==7==    at 0x75B846: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) [clone .isra.59] (in /home/jwnimmer/.cache/bazel/_bazel_jwnimmer/8d1e0fabf4d48d295f3b4d143e04c94c/execroot/drake-distro/bazel-out/gcc-4.9-linux-dbg/bin/drake/solvers/nonlinear_program_test)
==7==    by 0x75D95E: Ipopt::RegisteredOptions::GetOption(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /home/jwnimmer/.cache/bazel/_bazel_jwnimmer/8d1e0fabf4d48d295f3b4d143e04c94c/execroot/drake-distro/bazel-out/gcc-4.9-linux-dbg/bin/drake/solvers/nonlinear_program_test)
==7==    by 0x758653: Ipopt::OptionsList::SetNumericValue(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, bool, bool) (in /home/jwnimmer/.cache/bazel/_bazel_jwnimmer/8d1e0fabf4d48d295f3b4d143e04c94c/execroot/drake-distro/bazel-out/gcc-4.9-linux-dbg/bin/drake/solvers/nonlinear_program_test)
==7==    by 0x62E958: drake::solvers::IpoptSolver::Solve(drake::solvers::MathematicalProgram&) const (ipopt_solver.cc:493)
```
... followed later by a segfault.

Note that this is not a supported configuration.  Before testing, we must `apt install g++-4.9`.

jwnimmer-tri, 2017-05-15:
FYI @m-chaturvedi @jamiesnape in case you have already come across or fixed this, you could update the ticket here.  If you don't have any info, that's fine and you can ignore this ticket.

m-chaturvedi, 2017-05-15:
David had pointed out that COW is happening on gcc 4.9 on Slack.  Perhaps the issue has something to do with it.  Here's a gist which maybe interesting:
https://gist.github.com/m-chaturvedi/1262b200d26129512ecf92fa6e2fc4e3

sammy-tri, 2018-09-21:
I don't believe drake will build against gcc 4.9 anymore.

closed 2018-09-21 by sammy-tri