RobotLocomotion/drake #5511: Wrong namespace in drake/multibody code

jwnimmer-tri, 2017-03-16:
Approximately 49 files need a repair to their namespace.  I feel like doing this sooner rather than later will make everyone's life easier:

```
$ find drake/multibody -type f -print0 | xargs -0 grep -n -e 'namespace systems'
drake/multibody/constraint_wrappers.cc:4:namespace systems {
drake/multibody/constraint_wrappers.cc:107:}  // namespace systems
drake/multibody/inverse_kinematics_backend.h:22:namespace systems {
drake/multibody/inverse_kinematics_backend.h:92:}  // namespace systems
drake/multibody/ik_trajectory_helper.h:10:namespace systems {
drake/multibody/ik_trajectory_helper.h:115:}  // namespace systems
drake/multibody/rigid_body_plant/rigid_body_plant_that_publishes_xdot.h:18:namespace systems {
drake/multibody/rigid_body_plant/rigid_body_plant_that_publishes_xdot.h:66:}  // namespace systems
drake/multibody/rigid_body_plant/contact_detail.cc:4:namespace systems {
drake/multibody/rigid_body_plant/contact_detail.cc:15:}  // namespace systems
drake/multibody/rigid_body_plant/viewer_draw_translator.h:13:namespace systems {
drake/multibody/rigid_body_plant/viewer_draw_translator.h:59:}  // namespace systems
drake/multibody/rigid_body_plant/contact_resultant_force_calculator.h:11:namespace systems {
drake/multibody/rigid_body_plant/contact_resultant_force_calculator.h:258:}  // namespace systems
drake/multibody/rigid_body_plant/drake_visualizer.h:19:namespace systems {
drake/multibody/rigid_body_plant/drake_visualizer.h:145:}  // namespace systems
drake/multibody/rigid_body_plant/contact_info.h:12:namespace systems {
drake/multibody/rigid_body_plant/contact_info.h:89:}  // namespace systems
drake/multibody/rigid_body_plant/rigid_body_plant_that_publishes_xdot.cc:10:namespace systems {
drake/multibody/rigid_body_plant/rigid_body_plant_that_publishes_xdot.cc:64:}  // namespace systems
drake/multibody/rigid_body_plant/viewer_draw_translator.cc:12:namespace systems {
drake/multibody/rigid_body_plant/viewer_draw_translator.cc:84:}  // namespace systems
drake/multibody/rigid_body_plant/contact_force.h:7:namespace systems {
drake/multibody/rigid_body_plant/contact_force.h:114:}  // namespace systems
drake/multibody/rigid_body_plant/kinematics_results.cc:8:namespace systems {
drake/multibody/rigid_body_plant/kinematics_results.cc:118:}  // namespace systems
drake/multibody/rigid_body_plant/kinematics_results.h:10:namespace systems {
drake/multibody/rigid_body_plant/kinematics_results.h:93:}  // namespace systems
drake/multibody/rigid_body_plant/rigid_body_plant.cc:26:namespace systems {
drake/multibody/rigid_body_plant/rigid_body_plant.cc:927:}  // namespace systems
drake/multibody/rigid_body_plant/point_contact_detail.h:9:namespace systems {
drake/multibody/rigid_body_plant/point_contact_detail.h:33:}  // namespace systems
drake/multibody/rigid_body_plant/contact_resultant_force_calculator.cc:7:namespace systems {
drake/multibody/rigid_body_plant/contact_resultant_force_calculator.cc:207:}  // namespace systems
drake/multibody/rigid_body_plant/contact_force.cc:8:namespace systems {
drake/multibody/rigid_body_plant/contact_force.cc:46:}  // namespace systems
drake/multibody/rigid_body_plant/contact_results_to_lcm.h:11:namespace systems {
drake/multibody/rigid_body_plant/contact_results_to_lcm.h:38:}  // namespace systems
drake/multibody/rigid_body_plant/contact_detail.h:9:namespace systems {
drake/multibody/rigid_body_plant/contact_detail.h:43:}  // namespace systems
drake/multibody/rigid_body_plant/rigid_body_plant.h:17:namespace systems {
drake/multibody/rigid_body_plant/rigid_body_plant.h:459:}  // namespace systems
drake/multibody/rigid_body_plant/contact_results.cc:6:namespace systems {
drake/multibody/rigid_body_plant/contact_results.cc:38:}  // namespace systems
drake/multibody/rigid_body_plant/contact_results_to_lcm.cc:9:namespace systems {
drake/multibody/rigid_body_plant/contact_results_to_lcm.cc:55:}  // namespace systems
drake/multibody/rigid_body_plant/drake_visualizer.cc:10:namespace systems {
drake/multibody/rigid_body_plant/drake_visualizer.cc:172:}  // namespace systems
drake/multibody/rigid_body_plant/contact_results.h:10:namespace systems {
drake/multibody/rigid_body_plant/contact_results.h:54:}  // namespace systems
drake/multibody/rigid_body_plant/contact_info.cc:6:namespace systems {
drake/multibody/rigid_body_plant/contact_info.cc:49:}  // namespace systems
drake/multibody/rigid_body_plant/test/compute_contact_result_test.cc:33:namespace systems {
drake/multibody/rigid_body_plant/test/compute_contact_result_test.cc:190:}  // namespace systems
drake/multibody/rigid_body_plant/test/contact_info_test.cc:12:namespace systems {
drake/multibody/rigid_body_plant/test/contact_info_test.cc:82:}  // namespace systems
drake/multibody/rigid_body_plant/test/contact_force_formula_test.cc:15:namespace systems {
drake/multibody/rigid_body_plant/test/contact_force_formula_test.cc:433:}  // namespace systems
drake/multibody/rigid_body_plant/test/drake_visualizer_test.cc:18:namespace systems {
drake/multibody/rigid_body_plant/test/drake_visualizer_test.cc:526:}  // namespace systems
drake/multibody/rigid_body_plant/test/kinematics_results_test.cc:18:namespace systems {
drake/multibody/rigid_body_plant/test/kinematics_results_test.cc:118:}  // namespace systems
drake/multibody/rigid_body_plant/test/rigid_body_plant_test.cc:35:namespace systems {
drake/multibody/rigid_body_plant/test/rigid_body_plant_test.cc:600:}  // namespace systems
drake/multibody/rigid_body_plant/test/contact_resultant_force_test.cc:13:namespace systems {
drake/multibody/rigid_body_plant/test/contact_resultant_force_test.cc:454:}  // namespace systems
drake/multibody/rigid_body_plant/test/contact_detail_test.cc:12:namespace systems {
drake/multibody/rigid_body_plant/test/contact_detail_test.cc:39:}  // namespace systems
drake/multibody/rigid_body_plant/test/viewer_draw_translator_test.cc:16:namespace systems {
drake/multibody/rigid_body_plant/test/viewer_draw_translator_test.cc:119:}  // namespace systems
drake/multibody/rigid_body_plant/test/contact_force_test.cc:8:namespace systems {
drake/multibody/rigid_body_plant/test/contact_force_test.cc:64:}  // namespace systems
drake/multibody/rigid_body_plant/point_contact_detail.cc:4:namespace systems {
drake/multibody/rigid_body_plant/point_contact_detail.cc:21:}  // namespace systems
drake/multibody/ik_trajectory_helper.cc:11:namespace systems {
drake/multibody/ik_trajectory_helper.cc:320:}  // namespace systems
drake/multibody/inverse_kinematics_trajectory_backend.cc:31:namespace systems {
drake/multibody/inverse_kinematics_trajectory_backend.cc:583:}  // namespace systems
drake/multibody/joints/test/joint_test.cc:23:namespace systems {
drake/multibody/joints/test/joint_test.cc:138:}  // namespace systems
drake/multibody/constraint_wrappers.h:18:namespace systems {
drake/multibody/constraint_wrappers.h:86:}  // namespace systems
drake/multibody/inverse_kinematics_backend.cc:40:namespace systems {
drake/multibody/inverse_kinematics_backend.cc:332:}  // namespace systems
drake/multibody/test/rigid_body_tree/rigid_body_tree_clone_test.cc:35:namespace systems {
drake/multibody/test/rigid_body_tree/rigid_body_tree_clone_test.cc:208:}  // namespace systems
drake/multibody/test/rigid_body_tree/rigid_body_collision_clique_test.cc:14:namespace systems {
drake/multibody/test/rigid_body_tree/rigid_body_collision_clique_test.cc:154:}  // namespace systems
drake/multibody/test/rigid_body_tree/rigid_body_tree_face_extraction_test.cc:15:namespace systems {
drake/multibody/test/rigid_body_tree/rigid_body_tree_face_extraction_test.cc:70:}  // namespace systems
drake/multibody/test/rigid_body_tree/rbt_collisions_test.cc:15:namespace systems {
drake/multibody/test/rigid_body_tree/rbt_collisions_test.cc:222:}  // namespace systems
drake/multibody/test/rigid_body_tree/rigid_body_tree_creation_test.cc:16:namespace systems {
drake/multibody/test/rigid_body_tree/rigid_body_tree_creation_test.cc:440:}  // namespace systems
drake/multibody/test/rigid_body_tree/rigid_body_tree_dynamics_test.cc:28:namespace systems {
drake/multibody/test/rigid_body_tree/rigid_body_tree_dynamics_test.cc:350:}  // namespace systems
drake/multibody/test/rigid_body_test.cc:16:namespace systems {
drake/multibody/test/rigid_body_test.cc:112:}  // namespace systems
```

sherm1, 2017-03-16:
This doesn't seem urgent to me given that this old code has been sitting there that way for ages with no noticeable symptoms. But ... if anyone is looking for some occupational therapy and would like the pleasure of an easy-to-land PR, please fix!

jwnimmer-tri, 2017-03-16:
The major pain point is that `RigidBodyPlant` (and its helper classes for abstract output port data) are in the wrong namespace.  RBP is a central element of Drake's API; by having it in a namespace that is doomed to change, as more and more using-code gets written the impact of doing the rename later goes up.  Right now we have a few major in-tree users (each with multiple touchpoints to RBP) under active development, and a couple downstream separate-codebase users also.  To me it seems worth fixing this sooner rather than later; it will only get harder over time.

sherm1, 2017-03-16:
OK. @amcastro-tri, would you please fix the namespace when you have a chance?

jwnimmer-tri, 2017-06-19:
Re-ping.

jwnimmer-tri, 2018-12-21:
This issue is about the RBT code, in the attic. We're not going to fix it, it's okay to leave it however it is now.

closed 2018-12-21 by jwnimmer-tri