RobotLocomotion/drake #5520: Cannot construct Atlas model using pydrake bindings

rdeits, 2017-03-16:
```python
import pydrake.rbtree as rbt
robot = rbt.RigidBodyTree("/home/rdeits/locomotion/drake-distro/drake/examples/Atlas/urdf/atlas_minimal_contact.urdf")
```

gives:

```
RuntimeError: /home/rdeits/locomotion/explorations/drake-mpc/build/drake-prefix/src/drake/drake/multibody/parsers/urdf_parser.cc: ParseGeometry: ERROR: Mesh file name could not be resolved from the provided uri "package://Atlas/urdf/meshes/r_clav.obj".
```

It looks like that constructor needs to be updated to use the package path. I'll look into it.

liangfok, 2017-03-16:
See `drake/multibody/parsers/package_map.h`, specifically methods `PopulateFromFolder()`,  `PopulateFromEnvironment()`, and `PopulateUpstreamToDrake()`, which might be helpful towards the resolution of this issue.

rdeits, 2017-03-16:
Yup, thanks @liangfok . That's all working; I just have to figure out how to deal with the way non-copyable types play with pybind11.

PR #5523 (2017-03-16): Bind PackageMap fully in Python and update RBT python constructor
Fixes #5520

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5523)
closed 2017-03-16 by rdeits