RobotLocomotion/drake #5541: AutomotiveSimulator Design That Includes PoseAggregator

liangfok, 2017-03-17:
This issue tracks the redesign of `AutomotiveSimualtor` to support the `PoseAggregator`. It's also a solicitation for feedback comments.

# Current Design
The current design is shown in the image below. Multiple levels of multiplexers are used to stitch together `x_c`, which is then fed into `DrakeVisualizer`.

[image]

# Proposed Redesign (Version 1)

Each vehicle model instance outputs a `PoseVector` and `FrameVelocity` that describes the pose and time-derivative-of-the-pose of its "root" body. For each model instance, its outputs are fed into a `CarVisualizer` that adds the poses of all of the other bodies that collectively represent the visualization of the model instance. It outputs a `PoseBundle`. All of these `PoseBundle` outputs are then fed into the `PoseAggreagator`, whose output is fed into a `PoseBundleToDrawMesssge` system. Note that each `CarVisualizer` system also outputs an `lcmt_viewer_load_robot` message that is then also sent to the visualizer.

[image]

david-german-tri, 2017-03-17:
First pass, two thoughts:

1. Did you consider producing and sending the `lcmt_viewer_load_robot` message outside of the System diagram?  It only needs to be sent once, and everything you need to know to send it is (I think) available before the simulation begins.  Then you can drop the N `LcmPublisherSystem`s.  For an example of this approach, see https://gist.github.com/david-german-tri/0021ef9ff01602e12bf0776b95a7413e

2. I like the `CarVisualization` approach to fixed dependent transforms for now, but it also illustrates the need for a dependent transform primitive in the load-robot message in the future.

SeanCurtis-TRI, 2017-03-17:
@sherm1 and I are operating under the assumption that the system will eventually support an "initialize" phase.  I would hope the `lcmt_viewer_load_robot` messages would be resolved there, explicitly.

jwnimmer-tri, 2017-03-17:
First, this seems like a duplicate of #4112.
Second, David is writing a big reply and I'll wait for that before posting more.

david-german-tri, 2017-03-17:
> @sherm1 and I are operating under the assumption that the system will eventually support an "initialize" phase

I think this would be a misfeature and we should not do it.  (@sherm1 has heard me say so many times before, but @SeanCurtis-TRI may not have.)  The System framework's mission is to express and compose hybrid models.  The dataflow architecture is geared towards answering the question "given this state, what is the next state?" in a form that's safe, transparent, and consistent with the controls literature.

Initialization is not the same problem, or even much a related one.  If there's procedural one-time stuff you need to do before commencing simulation, C++ itself is a great way to express procedural stuff.  The OP is an excellent example.  To make it correct as drawn, yes, you need an initialization phase (or a mutable state hack that simulates an initialization phase).  But it is vastly simpler and cleaner not to think of `load_robot` publication as a Diagram activity at all.

SeanCurtis-TRI, 2017-03-17:
So, it seems to my that `System ⊂ Simulator`.  While I can see the point you make about the purpose of the `System`, simulators benefit from initialization.  And, in fact, it seems that it could rear its head when we actually move from one continuous interval to another (separated by discrete events which modify, or even rebuild the system.)

Furthermore, the mission of the `System` has obviously grown beyond your definition, otherwise there would be no room for `Publish`.

Moral of the story, I'm open to arguments for and against `Initialization`.  I just don't think it's a foregone conclusion.

jwnimmer-tri, 2017-03-17:
It's possible that `CarVisualization` be a bus-to-bus system (for all cars) rather that just one car would scale better in code complexity and runtime complexity.  Especially when we add world geometry into the mix, such as the maliput meshes or flat planes.  Liang, we can chat live on this when you have time.

liangfok, 2017-03-17:
Thanks for all of the feedback! Let me ponder this some more over the weekend. Perhaps we can discuss F2F early next week.

PR #5664 (2017-03-30): Completes Migration of AutomotiveSimulator Architecture
Resolves #5541.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5664)
closed 2017-03-30 by liangfok