RobotLocomotion/drake #5562: Missing library errors when trying to launch drake-visualizer on OSX

rdeits, 2017-03-19:
I'm getting an error similar to #5081, but the workaround of running `install/bin/drake-visualizer` doesn't make a difference in my case:

```bash
± ./install/bin/drake-visualizer 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/python2.7/dist-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/python2.7/dist-packages/director/mainwindowapp.py", line 3, in <module>
    from director import consoleapp
  File "/Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/python2.7/dist-packages/director/consoleapp.py", line 6, in <module>
    from director import applogic
  File "/Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/python2.7/dist-packages/director/applogic.py", line 5, in <module>
    import director.vtkAll as vtk
  File "/Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/python2.7/dist-packages/director/vtkAll.py", line 2, in <module>
    from vtkDRCFiltersPython import *
ImportError: dlopen(/Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/python2.7/dist-packages/director/vtkDRCFiltersPython.so, 2): Library not loaded: liblcmtypes_bot2-core.dylib
  Referenced from: /Users/rdeits/locomotion/explorations/drake-mpc/build/install/lib/libbot2-frames.2.dylib
  Reason: image not found
```

(`find` shows that `liblcmtypes_bot2-core` is present in `install/lib/`) 

If I build Director standalone, I do not get this issue, and I can launch `drake-visualizer` without any problems. This only seems to happen when Director is built through the Drake superbuild on OSX. 

My current workaround is just to always have a standalone build of Director in addition to the entire Drake build. 

@patmarion I have a vague recollection that this is due to the duplication of `bot_core` in `site-packages` and `dist-packages`. Does that make sense?

patmarion, 2017-03-19:
https://github.com/RobotLocomotion/drake/pull/4830#issuecomment-282929147

rdeits, 2017-03-19:
Ah, yes, thanks. Sorry, I somehow failed to find that issue in my searching.

patmarion, 2017-04-07:
Fixed by #4830

closed 2017-04-07 by patmarion