RobotLocomotion/drake #5564: Create and host VTK binary packages for Drake's supported platforms

jwnimmer-tri, 2017-03-19:
Sub-task broken out of #2628:
> Create vtk 6.x and 7.1 binary packages for Mac, Ubuntu 14.04, and Ubuntu 16.04, host them somewhere, and build enough CMake infrastructure so that the superbuilds can download and link against them. This is not a PPA, just a dpkg or binary tarball.

I would like to get the 7.1 packages up earlier rather than later, because they will be the long pole for Drake-as-a-library to be 100% compile-able under Bazel -- because our `drake/systems/sensors/rgbd_camera` component needs VTK to compile.

jwnimmer-tri, 2017-03-20:
At https://github.com/jwnimmer-tri/drake/tree/bazel-vtk, I've posted work-in-progress to adding VTK7 to the Bazel workspace.  It's only on Xenial right now, using the pre-compiled https://launchpad.net/~elvstone/+archive/ubuntu/vtk7 release.  Hopefully this can serve as a starting point to help evaluate suitability of Kitware's hosted VTK7 binaries for Xenial, Trusty, OS X, etc.

The `drake/systems/sensors:vtk_util_test` passes.  The `drake/systems/sensors:rgbd_camera_test` passes except for `ImageTest.MeshBoxRenderingTest`, which is somehow failing to put the green `png` texture into effect.  (There is also no `$DISPLAY` in the test sandbox right now, so the `rgbd_camera_test` has to be run by hand.)

david-german-tri, 2017-03-20:
@mwoehlke-kitware is working on this for Linux, and @jamiesnape for Mac.

jwnimmer-tri, 2017-03-23:
FYI I've updated my https://github.com/jwnimmer-tri/drake/tree/bazel-vtk branch to use the new tarballs.

jwnimmer-tri, 2017-03-25:
Once we're using the prebuilts, we should remove any unnecessary VTK-related instructions and packages from the `from_source` docs and `setup/16.04/...` script.

david-german-tri, 2017-04-12:
@sankhesh et. al. - I've heard periodically over the past couple weeks that the packages are done.  Could you PR documentation updates about where to find them, and then close this issue?  Thanks!

jamiesnape, 2017-04-12:
@mwoehlke-kitware @sankhesh:

Linux:

https://d2mbb5ninhlpdu.cloudfront.net/vtk/vtk-6.3.0-trusty-x86_64.tar.xz
https://d2mbb5ninhlpdu.cloudfront.net/vtk/vtk-6.3.0-xenial-x86_64.tar.xz
https://d2mbb5ninhlpdu.cloudfront.net/vtk/vtk-7.1.0-trusty-x86_64.tar.xz
https://d2mbb5ninhlpdu.cloudfront.net/vtk/vtk-7.1.0-xenial-x86_64.tar.xz

Mac:
```
brew tap robotlocomotion/director
brew install vtk@6.3
brew install vtk@7.1
```

jamiesnape, 2017-04-12:
S3 buckets are `s3://drake-packages` and `s3://drake-homebrew`.

sankhesh, 2017-04-12:
@jamiesnape @mwoehlke-kitware  Is this the page that needs to be updated: http://drake.mit.edu/from_binary.html (https://github.com/RobotLocomotion/RobotLocomotion.github.io/blob/master/from_binary.html)?

jamiesnape, 2017-04-12:
Good question. For now, I think I would create a separate vtk page since we are not fully integrated with the build. Does that sound good @david-german-tri?

david-german-tri, 2017-04-12:
> For now, I think I would create a separate vtk page since we are not fully integrated with the build. Does that sound good @david-german-tri?

Agree

david-german-tri, 2017-04-20:
Also, the tools/scripts for building these packages should be checked in somewhere.

mwoehlke-kitware, 2017-04-20:
On 2017-04-20 10:41, David German wrote:
> Also, the tools/scripts for building these packages should be
> checked in somewhere.

The build script is at
https://console.aws.amazon.com/s3/buckets/drake-provisioning/vtk/vtk-build.sh/details.
Does this seem like a reasonable place to keep it (right next to the
binaries themselves)?

-- 
Matthew

jwnimmer-tri, 2017-04-20:
(1) That URL is broken for me (404).

(2) It seems like source control would serve us better (the bucket could have a pointer to the source repository).  I think putting it into Drake's git would be great.

jamiesnape, 2017-04-20:
> (1) That URL is broken for me (404).

It should be. Not sure why it was posted here.

> (2) It seems like source control would serve us better (the bucket could have a pointer to the source repository). I think putting it into Drake's git would be great.

Agreed.

mwoehlke-kitware, 2017-04-20:
>> (1) That URL is broken for me (404).
>
> It should be. Not sure why it was posted here.

...because I don't know the magic incantation to turn it into a link that lets anyone download the file?

stonier, 2017-05-18:
Just for the record, these will be:

* Qt4/VTK8 on Trusty, Qt5/VTK8 on Xenial/Mac

jamiesnape, 2017-05-18:
Bazel integration is in progress. I almost have the VTK parts of drake sensors building, Director will be a little longer.

sankhesh, 2017-06-01:
@jamiesnape @stonier @patmarion 

## Pre-compiled binaries for `vtk` and `director`
- Ubuntu 14.04 (Trusty Tahr)
    - **VTK** (master@[28deb56](https://gitlab.kitware.com/vtk/vtk/commits/28deb56))
    [vtk-v7.1.1-1584-g28deb56-qt-4.8.6-trusty-x86_64.tar.gz](https://d2mbb5ninhlpdu.cloudfront.net/vtk/vtk-v7.1.1-1584-g28deb56-qt-4.8.6-trusty-x86_64.tar.gz)
    - **director**
    [dd-0.1.0-133-gd53b223-qt-4.8.6-trusty-x86_64.tar.gz](https://d2mbb5ninhlpdu.cloudfront.net/director/dd-0.1.0-133-gd53b223-qt-4.8.6-trusty-x86_64.tar.gz)
- Ubuntu 16.04 (Xenial Xerus)
    - **VTK** (master@[28deb56](https://gitlab.kitware.com/vtk/vtk/commits/28deb56))
    [vtk-v7.1.1-1584-g28deb56-qt-5.5.1-xenial-x86_64.tar.gz](https://d2mbb5ninhlpdu.cloudfront.net/vtk/vtk-v7.1.1-1584-g28deb56-qt-5.5.1-xenial-x86_64.tar.gz)
    - **director**
    [dd-0.1.0-133-gd53b223-qt-5.5.1-xenial-x86_64.tar.gz](https://d2mbb5ninhlpdu.cloudfront.net/director/dd-0.1.0-133-gd53b223-qt-5.5.1-xenial-x86_64.tar.gz)
- Mac OS X / macOS
    - VTK and Qt
    ```sh
    $ brew tap robotlocomotion/director
    $ brew install qt@5 vtk@8
    ```
    - **director**
    [dd-0.1.0-133-gd53b2231-qt-5.8.0-Darwin.tar.gz](https://d2mbb5ninhlpdu.cloudfront.net/director/dd-0.1.0-133-gd53b2231-qt-5.8.0-Darwin.tar.gz)

stonier, 2017-06-06:
@jamiesnape assigning you to cover the docs part as we discussed last meeting. 

* [ ] make sure `brew install qt@5 vtk@8` is working (@sankhesh reported problems over in director)
* [ ] hosting information and vtk8 build instructions to go into the CI docs maintained by kitware
* [ ] update the `from_source` docs here relevant to vtk
* [ ] drop vtk from the setup/ubuntu/16.04 script

jamiesnape, 2017-06-06:
> (@sankhesh reported problems over in director)

Indeed. Looking into it. Not sure why. It works perfectly on three different Macs for me.

EricCousineau-TRI, 2018-02-08:
Closing, as this appears to have been completed.

closed 2018-02-08 by EricCousineau-TRI