RobotLocomotion/drake #5566: Windows- error :  bot2-lcmgl.jar cannot be found by drake

hughhugh, 2017-03-20:
Hi,
I use the release binary version of drake for windows 64. I have run several examples, it works. The checkdependcy about lcmgl didnot succeecd when it came  to the simulation https://github.com/andybarry/simflight . I have searched and found that it is the issue of drake.  So is it possible to get this issue solved? Thank you.

RussTedrake, 2017-03-20:
I'm afraid that we're not supporting windows anymore.  lcmgl is only a debug plotting tool, i think the best solution for you is to comment out the relevant calls (starting with the "checkdependency")

closed 2017-03-20 by RussTedrake

hughhugh, 2017-03-20:
Hi，thank you. I will have a try to comment the checkdependency to see if it works.