RobotLocomotion/drake #5568: Overhead when calling Mosek through MathematicalProgram

rdeits, 2017-03-20:
@TobiaMarcucci and I noticed that solving a trivial LP with Mosek through MathematicalProgram is surprisingly slow. The LP is:

```
min. 0 
s. t. 0 <= x_i <= 1  for i in {1, 2}
```

In python, this looks like:

```python
prog = mp.MathematicalProgram()
x = prog.NewContinuousVariables(2, "x")
prog.AddLinearConstraint(x[0] >= 0)
prog.AddLinearConstraint(x[0] <= 1)
prog.AddLinearConstraint(x[1] >= 0)
prog.AddLinearConstraint(x[1] <= 1)

solver = MosekSolver()
tic = time.time()
solver.Solve(prog)
print time.time() - tic
```

Solving that program takes 10-15 milliseconds on my laptop (OSX 10.11). In comparison, using Gurobi (via the `GurobiSolver`) takes 0.5-1 ms. 

This result is not specific to this particular LP; it's just that the overhead is much more apparent for very simple LPs like this one. 

I don't think the problem is actually with Mosek itself: solving the same LP without MathematicalProgram (from Julia) I get 0.5-1ms for both Mosek and Gurobi. It looks like MathematicalProgram has a lot of overhead when communicating with Mosek. 

@hongkai-dai this is not a high priority. We're not planning on using Mosek anyway. But I thought it would be good to write this down.

hongkai-dai, 2017-03-20:
In `mosek_solver.cc`, I am not parsing the problem efficiently. I will run calgrind to profile the problem. Thanks for doing the comparison.

hongkai-dai, 2017-03-20:
And I am glad that `MathematicalProgram` and Julia takes about the same time for the gurobi solver.

hongkai-dai, 2019-03-06:
Sorry for not working on this for a long time. I just created the same test in C++. Now this program takes 1.2ms to call `mosek_solver.Solve(prog, {}, {}, &result)`. Out of that 1.2 ms, Mosek spends 1.03ms to actually solve the problem, the rest are the overhead. So I think the overhead is small.

My test is in https://github.com/hongkai-dai/drake/blob/mosek_lp_profile/solvers/test/mosek_lp_profile.cc

hongkai-dai, 2019-03-19:
@TobiaMarcucci could you rerun the test in https://github.com/RobotLocomotion/drake/issues/5568#issue-215466286 on your machine? I cannot test it on mine because first I don't have a mac machine, and second we don't have a standalone license for TRI, so calling Mosek::Solve would incur contacting a TRI license server, and the time for probing that server is counted into the elapsed time when I execute that python script.

TobiaMarcucci, 2019-03-20:
@hongkai-dai I recompiled drake and tried i, but it still takes around 10 ms on average.

hongkai-dai, 2019-03-20:
Thanks @TobiaMarcucci for testing. Just to make sure that we are on the same page, is it OK if you could check out the branch https://github.com/hongkai-dai/drake/tree/mosek_lp_profile, and run the c++ test? The command is
```
$ bazel build --config mosek //solvers:mosek_lp_profile
$ ./bazel-bin/solvers/mosek_lp_profile
```
It should print out the total time to call MosekSolver::Solve() in the first line (including passing data to Mosek), and the actual Mosek solver time on the second line.

TobiaMarcucci, 2019-03-20:
I tried, but I got this error:
```
~/codes/drake$ bazel build --config mosek //solvers:mosek_lp_profile
Starting local Bazel server and connecting to it...
INFO: Invocation ID: 679edfab-bac6-467d-8a98-a6866954774b
INFO: Analysed target //solvers:mosek_lp_profile (32 packages loaded, 1099 targets configured).
INFO: Found 1 target...
ERROR: /Users/tobia/codes/drake/solvers/BUILD.bazel:1538:1: C++ compilation of rule '//solvers:mosek_lp_profile' failed (Exit 1) wrapped_clang failed: error executing command external/local_config_cc/wrapped_clang '-D_FORTIFY_SOURCE=1' -fstack-protector -fcolor-diagnostics -Wall -Wthread-safety -Wself-assign -fno-omit-frame-pointer -g0 -O2 '-D_FORTIFY_SOURCE=1' -DNDEBUG ... (remaining 237 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
solvers/test/mosek_lp_profile.cc:20:46: error: use of old-style cast [-Werror,-Wold-style-cast]
  double duration = (std::clock() - start) / (double)CLOCKS_PER_SEC;
                                             ^       ~~~~~~~~~~~~~~
1 error generated.
Target //solvers:mosek_lp_profile failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 68.585s, Critical Path: 23.25s
INFO: 80 processes: 80 darwin-sandbox.
FAILED: Build did NOT complete successfully
```
How can I solve it? Thanks!

hongkai-dai, 2019-03-20:
My bad, I just force push a new commit to the branch, could you compile and run it again?

(Since I did force push, you might need to remove the your local `mosek_lp_profile` branch, and then check out the branch from the remote again).

TobiaMarcucci, 2019-03-20:
Done. On average, I get something like:
```
time: 0.001116s
mosek solve time: 0.000275135s
```

hongkai-dai, 2019-03-20:
Thanks @TobiaMarcucci , that is interesting result
First the total time in the C++ code is about 1.1 ms on your machine, much less than the Python code with 10ms. Second the mosek solve time is just 0.3 ms. That's smaller than the time on my machine with XEON CPU (mine is about 1 ms). I will try to investigate into it.