RobotLocomotion/drake #5629: publish in Simulator.

siyuanfeng-tri, 2017-03-27:
I think the current mechanism for handling publishing events in the simulator needs to be improved. Here are a couple complaints. 
1. There is a global publish on every major time step flag in the simulator API, I think that should needs to be changed somehow. Imagine some system is declared to publish at 200hz, and simulator's major time step is 1ms, the message rate can be 1200hz depending on whether that flag is set to true or not. While intuitively, the rate should always be at 200hz regardless. 
2. I think publish on everytime step flag should be in the System API instead of the Simulator API. The current implementation doesn't handle situations where Subsys A wants to publish every major time step, and Subsys B doesn't.
3. I am not sure calling Publish in Simulator's Initialization() is a good idea. If it's just to force trigger something like this. https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/rigid_body_plant/drake_visualizer.cc#L119. Here is a example:
Imagine some subsystem that requires custom initialization, and authors put some asserts in the DoCalcOutput that checks for some initial condition being satisfied. In this case, you can't directly call simulator.Initiliaze() because that calls Publish which could trigger EvalInput calls which then triggers the DoCalcOutput and cause assertion fails. But you can't custom call custom initialization because the context hasn't being allocated. I suppose we can circumvent this by allocating that manually, and pass it to the simulator ctor... 

cc @edrumwri @sherm1 @jwnimmer @liangfok 
i am happy to work on this. soliciting inputs.

liangfok, 2017-03-27:
Regarding (3), perhaps we should modify `DrakeVisualizer` to provide a `get_load_message()` method, which returns an `lcmt_viewer_load_robot` object. The user of this system, which is typically the application, would then be responsible for sending the load message outside of the system framework's context. This would match what `CarVisApplicator` does: https://github.com/RobotLocomotion/drake/blob/f6c49a0459cc323a15ccabf9feefebe039d346d4/drake/automotive/car_vis_applicator.h#L70.

siyuanfeng-tri, 2017-03-27:
@liangfok i think this better.

sherm1, 2017-03-27:
I agree that "when to publish" should be determined by the System and individual sub-systems; I think that has always been the plan and at least part of that functionality is in place. Publishing events are supposed to provide the same control as other events like discrete update events -- every step, periodic, triggered, etc. @siyuanfeng-tri it would be great if you want to take a look at that.

sherm1, 2017-03-27:
First step would be to assess what's there now and propose what needs changing.

siyuanfeng-tri, 2017-03-27:
I think the current Publish() api call in System is more like a "force publish". Ideally, I would like to have finer control in each subsys to cover: periodic publish, publish on every major step, force publish, silent (never publish). I am debating whether it should publish on minor time steps inside the integrator. 

The obvious way to do this would to have a some flag and check for context.get_time in Publish() combined with the current event handling logic, which fires whenever any subsystem needs to publish.

siyuanfeng-tri, 2017-03-27:
forgot to cc @david-german-tri

sherm1, 2017-03-27:
> I am debating whether it should publish on minor time steps inside the integrator.

I don't think so -- that would only be for internal debugging and I would rather not do that through a normal API method.

sherm1, 2017-03-27:
@siyuanfeng-tri, see the [`DiscreteEvent`](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/framework/system.h#L36) struct used by System -- it already has publish events defined and likely has most of what you want at least conceived and I think mostly functional. It is possible to provide your own publish callback.

siyuanfeng-tri, 2017-03-27:
@sherm1 does the context / system know about major time step size? 
I basically want to add that to DoCalcNextUpdateTimeImpl() in leaf_system.h

sherm1, 2017-03-27:
No, and it isn't possible to add it to DoCalcNextUpdateTime() because it can't be predicted. In general the major time step is discovered by the integration method after error analysis of a step determines if it is acceptable.

So a System just needs to say whether it wants a call at the next major time step and if so the Simulator will take care of making sure that happens.

siyuanfeng-tri, 2017-03-27:
@sherm1 maybe if you have time, we could have a 10mins quick hangout?

edrumwri, 2017-03-27:
I'm happy to do this too- brain has been slow to boot today so I need to
check it before we would speak.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

david-german-tri, 2017-04-03:
> So a System just needs to say whether it wants a call at the next major time step and if so the Simulator will take care of making sure that happens.

Do we really want this feature?  Integrator timesteps are by nature arbitrary; it's hard for me to understand what a System that publishes on integrator timesteps would be modeling.  (The default `Publish`-on-every-timestep behavior is for legacy compatibility.  New code shouldn't use it, and sooner or later we'll delete it, I hope.)

That said, if we do want this feature, I agree it should be expressed as a new field in `DiscreteEvent`.

siyuanfeng-tri, 2017-04-03:
@david-german-tri it's definitely useful for debugging purposes. especially some logging tools seem to rely on Publish for logging. There are also fixed step integrator.

I really want to publish on every major time step, but the current thing is incorrect, and it really confuses people.

sherm1, 2017-04-03:
>  Integrator timesteps are by nature arbitrary

They aren't actually arbitrary because the step size of a variable step integrator is chosen based (roughly) on the frequency content of the function being integrated, which is highly variable. So the major time steps are spaced evenly in "frequency space" and are the right times to do discrete updates that should have content-dependent update rates. Two simple examples: (1) tracking of min/max values, (2) updating a delay table that will later be interpolated for past values.

david-german-tri, 2017-04-03:
Yes, sorry, "arbitrary" was a poor choice of words.  What I meant was that the integrator step size depends on factors outside any given System's control or knowledge, such as the frequency content of other Systems that happen to appear in the same Diagram.

sherm1, 2017-04-03:
That's right but from an accuracy perspective the step size still provides a "safe" sampling rate so that all the interesting information in the simulation gets captured by the samples. There may be too many samples taken but that is better than too few and the best you can do if you don't have a priori knowledge about when samples should be taken. So the "major step size" trigger is very useful in practice, where such a priori info is in short supply.

david-german-tri, 2017-04-03:
> So the "major step size" trigger is very useful in practice, where such a priori info is in short supply.

That certainly makes sense for state changes (as in both your examples).  I guess the argument extends to publish events if you simply put a network interface in the middle of your System architecture, so I buy it.  Thanks for the explanation!

sherm1, 2019-01-18:
As of #10428 (and including a long series of previous Event improvements) this issue is resolved! The Simulator no longer publishes per-step unless you ask it to, and individual Systems can control whether they get per-step events.

closed 2019-01-18 by sherm1