RobotLocomotion/drake #5645: Add a special formula for positive semidefinite

hongkai-dai, 2017-03-28:
This formula takes an Eigen::Matrix<symbolic::Expression>. We should be able to write
```C++
Eigen::Matrix<symbolic::Expression, Eigen::Dynamic, Eigen::Dynamic> X(n, n);
Eigen::Matrix<symbolic::Expression, Eigen::Dynamic, Eigen::Dynamic> Y(n, n);
auto formula = (X >= Y);  // Returns a symbolic::Formula that (X - Y) is positive semidefinite.
auto formula = (X >= 0);  // Retuns a symbolic::Formula that X is positive semidefinite.
```
This should also work for static-sized matrix. We should possibly check if the matrix is square in the release mode, and if it is symmetric in the debug mode, if the checking is expensive.

hongkai-dai, 2017-03-28:
/cc @RussTedrake  
/cc @EricCousineau-TRI

soonho-tri, 2017-03-31:
I had a discussion with @jwnimmer-tri on this topic. We agreed that it's better to have a function:

```c++
Formula positive_semidefinite(const Eigen::Ref<const MatrixX<symbolic>>& M);
```

than to provide operator overloading for `>=` / `<=`. Here, our proposal is to write:

 - `positive_semidefinite(M)` instead of `M >= 0`.
 - `positive_semidefinite(M1 - M2)` instead of `M1 >= M2`.

If `positive_semidefinite` is too long, I think we can add a shorter one, i.e. `psd(M)` or `PSD(M)`.

---

The rationale behind this opinion are:
1. In the previous proposal, the meaning of `M1 >= M2` depends on the size of `M1` and `M2`. If they are square matrices, `M1 >= M2` denotes `psd(M1 - M2)`. Otherwise `M1 >= M2` means `⋀ᵢⱼ M1(i,j) >= M2(i, j)`. We think we should avoid this ambiguity.
2. Assuming that we allow a shorter function-name, writing `psd(M1 - M2)` is not so much harder than `M1 >= M2` while it's self-evident which function is invoked if we write `psd(M)`.

hongkai-dai, 2017-03-31:
Hmm, if we are going to create `Formula positive_semidefinite`, then I guess we already have something similar in `MathematicalProgram`, as
```C++
MathematicalProgram::AddPositiveSemidefiniteConstraint(M);
MathematicalProgram::AddPositiveSemidefiniteConstraint(M1 - M2);
```
as in https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/mathematical_program.h#L1924~L1960. So we do not really need `Formula positive_semidefinite`.

I think the reason why we want the operator overload, is for readability. `AddConstraint(M>=0)` is more readable than `AddPositiveSemidefiniteConstraint(M)`.

I think in the previous proposal, if `M1` and `M2` are Eigen::Matrix, then `M1 >= M2` ALWAYS mean positive semidefinite. We then check if `M1` and `M2` are square matrix. If not, it throws a runtime error. 
On the other hand, if `M1` and `M2` are Eigen::Array, then `M1 >= M2` means `⋀ᵢⱼ M1(i,j) >= M2(i, j)`.

soonho-tri, 2017-03-31:
> AddConstraint(M>=0) is more readable than AddPositiveSemidefiniteConstraint(M).

FYI, the new proposal allows us to write `AddConstraint(psd(M))` so that we can still use a generic `AddConstraint` function.

soonho-tri, 2017-03-31:
> I think in the previous proposal, if M1 and M2 are Eigen::Matrix, then M1 >= M2 ALWAYS mean positive semidefinite. We then check if M1 and M2 are square matrix. If not, it throws a runtime error.

For now, we need to write `AddLinearConstraint((A*x).array() >= b.array())`. With this proposal, we can eliminate `.array()` and write `AddLinearConstraint(A*x >= b)`.

soonho-tri, 2017-03-31:
BTW, as I mentioned in Slack, I personally think that it's good to use `>>=` and `<<=` operators here to denote PSD constraints. But @jwnimmer-tri disagreed with me on this.

hongkai-dai, 2017-03-31:
> For now, we need to write AddLinearConstraint((A*x).array() >= b.array()). With this proposal, we can eliminate .array() and write AddLinearConstraint(A*x >= b).

I like that, and the `psd(M)` looks good to me also. 

@RussTedrake , if you do not object, then I think we will do what Soonho mentions above, to add `Formula psd`, and `M1 >= M2` means elementwise inequality, for Eigen::Matrix of `M1`, `M2`.

jwnimmer-tri, 2017-03-31:
With Soonho's discussion to me, I didn't understand that `>=` means _only_ PSD xor runtime_error.  I thought it meant PSD xor _some other math_.  If it has only 0..1 meaning, then my concerns are greatly reduced.

soonho-tri, 2017-03-31:
@jwnimmer-tri , we had multiple options in multiple dimensions (`Eigen::Array` vs `Eigen::Matrix` / PSD vs element-wise comparison). So Hongkai and I had different pictures, I think.

> RussTedrake , if you do not object, then I think we will do what Soonho mentions above, to add Formula psd, and M1 >= M2 means elementwise inequality, for Eigen::Matrix of M1, M2.

Thank you, @hongkai-dai !

RussTedrake, 2017-04-02:
no objections.  i think the `psd()` notation is concise, and I agree that it is the most clear.

PR #5717 (2017-04-07): Add symbolic formula representing positive-semidefinite constraint
Close #5645.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5717)
closed 2017-04-07 by soonho-tri