RobotLocomotion/drake #5646: Return Binding<T> for each AddXXXConstraint and AddXXXCost

hongkai-dai, 2017-03-28:
Currently we do not have a consistent return type, sometimes we return void, sometimes a shared pointer to a constraint, and sometimes a Binding object. We should change all the return types in AddXXXCost and AddXXXConstraint in `MathematicalProgram` to `Binding<T>`, where `T` is the type of the `Constraint`.

hongkai-dai, 2017-03-28:
/cc @soonho-tri 
/cc @RussTedrake

EricCousineau-TRI, 2017-04-03:
@soonho-tri and I were speaking about what the best return type would be for the `AddXXX{Cost,Constraint}` functions would be. It appears that `Binding<T>` is itself an immutable view, which contains a `VectorXDecisionVariable` that would incur a list copy on each copy of itself. We already copy this once when pushing the registering the bindings in the corresponding `constraint_` or `cost_` lists; if we return a `Binding<T>`, we will then duplicate the copy constructor once more, versus if we return a `const Binding<T>&` pointing to the `back()` of the list, it will avoid an additional copy constructor.

@hongkai-dai What are your preferences on the matter? For now, I will start implementing this returning by value for consistency, and can change to a `const` reference later on, either in this issue or a separate issue.

soonho-tri, 2017-04-03:
> It appears that Binding<T> is itself an immutable view

This is a nice observation. I think it's safe to return `const Binding<T>&`.

hongkai-dai, 2017-04-03:
Great observation. I think returning `const Binding<T>&` is a good choice.

To make sure we returned a shared pointer to a constraint that is mutable, and we can update the constraint returned from `AddXXXConstraint`, can we test the following code?
```C++
MathematicalProgram prog;
auto x = prog.NewDecisionVariables<2>();
const Binding<LinearConstraint>& binding = prog.AddLinearConstraint(x(0) + x(1)<=2);
// Now we want to change the constraint stored in `binding`.
binding.constraint()->UpdateConstraint(Eigen::RowVector2d(1, 2), Vector1d(0), Vector1d(1));
EXPECT_TRUE(CompareMatrices(binding.constraint()->A(), EIgen::RowVector2d(1, 2)));
EXPECT_TRUE(CompareMatrices(binding.constraint()->lower_bound(), Vector1d(0)));
EXPECT_TRUE(CompareMatrices(binding.constraint()->upper_bound(), Vector1d(1)));
```

Returning `const Binding<T>` does mean that we cannot modify the bound variables stored in `binding.variables()`. For the moment I think it is fine. In the model predictive control, we do not always change the variables.

EricCousineau-TRI, 2017-04-03:
Can do! From looking at the implementation, it appears that this should be possible given that `constaint()` is a `shared_ptr<C>`, but I will definitely double-check this.

Just in case, I will first implement `Binding<T>`, and then try to make an atomic commit switching to `const Binding<T>` to make it (a little) less painful if we need to revert.

EricCousineau-TRI, 2017-04-03:
The only caveat I can think of at this point is if there needs to be implicit down-casting flexibility, e.g., `Binding<LinearConstraint>` to `Binding<Constraint>`, or something of that sort. I will see if I run into that issue when I try making the switch.

EricCousineau-TRI, 2017-04-03:
It appears that the down-casting caveat becomes a blocking issue, if we wish to have consistency in returning the specific constraint type. I have pushed my attempt here: [issue/5646_return_binding_downcast](https://github.com/EricCousineau-TRI/drake/commit/9412d3767fa39b1208e82f0e33e264f6ab0af061)

We can return `const` references if we decide to either:
a. Always return `const Binding<Constraint>&`, and force users to manually upcost (blech), or 
b. Always store a corresponding version of a `Binding<Constraint>` such that it can be returned as a reference.
Both of these options seem very dirty, so I would propose keeping the return type as a value, `Binding<T>`, rather than `const Binding<T>&`.

closed 2017-04-06 by EricCousineau-TRI

EricCousineau-TRI, 2017-04-13:
As a follow-up to above point, if we ever find that copying `Binding<C>` is too expensive, we can consider doing the following to maintain returning const rvalue references:

Introduce a base, `BindingBase`, which explicitly stores a `shared_ptr<Constraint> constraint_`.
In the sub-classed type, overload `constraint()` to return `shared_ptr<C>`, such that you can do a dynamic cast to maintain syntactic sugar of the sub-classed templated types.

Expanded:
```
class BindingBase {
public:
  ...
  shared_ptr<Constraint> constraint() const { return constraint_; }
private:
  shared_ptr<Constraint> constraint_;
};
...
template<typename C>
class Binding : public BindingBase {
public:
  shared_ptr<C> constraint() const { return dynamic_ptr_cast<C>(constraint_); }
};
...
// Then you can do:
const BindingBase& MathematicalProgram::AddConstraintViaDispatch(...) {
  if (linear)
    return AddLinearConstraint(...);
  if (quadratic)
    return AddQuadraticConstraint(...);
  // etc...
}
```

If possible, to maintain current setup a little more, can consider:
```
template<typename C>
class Binding;

template<>
class Binding<Constraint> {
  // Base implementation
}

template<typename C>
class Binding : public Binding<Constraint> {
  // Specialized implementation
};
```