RobotLocomotion/drake #5650: Make sure each type of constraint can be added in the symbolic form

hongkai-dai, 2017-03-28:
I believe currently the only left-over is the `AddBoundingBoxConstraint`.

hongkai-dai, 2017-03-28:
\cc @RussTedrake 
\cc @EricCousineau-TRI

soonho-tri, 2017-03-31:
@hongkai-dai : I'll do this today (just unassinged you).

hongkai-dai, 2017-03-31:
Thanks a lot! You may hold off the part on the polynomial constraint. I will send out a PR later to handle it. We currently do not have a `std::vector<Binding<PolynomialConstraint>>` stored inside `MathematicalProgram`. I think we should add that, with the proper test. But the change would be easier to review if it is in a separate PR. What do you think?

soonho-tri, 2017-03-31:
SGTM.

I'll add the following functions:
```c++
Binding<LinearEqualityConstraint> AddLinearEqualityConstraint(const symbolic::Formula& f);
Binding<BoundingBoxConstraint> AddBoundingBoxConstraint(const symbolic::Formula& f);
```
so that we can write `AddLinearEqualityConstraint(A*x == b)` and `AddBoundingBoxConstraint(A*x <= b)`.

For the latter, I first need to add `operator<=` between `Eigen::Matrix` of symbolic expressions.

closed 2018-02-01 by hongkai-dai