RobotLocomotion/drake #5651: Operator overloading between basic vectors / between basic vector and scalar

hongkai-dai, 2017-03-28:
\cc @RussTedrake 
\cc @EricCousineau-TRI

jwnimmer-tri, 2017-03-28:
This sounds scary.  Please elaborate?

hongkai-dai, 2017-03-28:
This is for adding constraints/costs in trajectory optimization, where we use basic vectors. Currently we have to convert basic vector to an Eigen vector, and then call `AddConstraint` with the Eigen vector. It would be more convenient if we can handle basic vector directly.

jwnimmer-tri, 2017-03-28:
Operator overloading can be difficult for readers to scrutinize what's happening.  Sometimes, the extra clarity of doing it longhand is worth the extra syntax.

Maybe link to some example code where you think the syntax baggage is too much?

hongkai-dai, 2017-03-28:
@RussTedrake , could you point the use case for converting BasicVector to Eigen vector in the trajectory optimization code?

sherm1, 2017-03-28:
> Operator overloading can be difficult for readers to scrutinize what's happening.

True, but if there is a case for operator overloading at all I think matrix-vector arithmetic is the place!

jwnimmer-tri, 2017-03-28:
Yup, I could definitely see it going either way here.  And we already have `op+=` and `op-=` overloaded, so maybe just need `op*=` or something.  But I think the burden of proof should be high, so we'll just have to see what the specific proposal is.

RussTedrake, 2017-03-28:
For example, I'm writing lots of trajectory optimization examples that look like this:
```
  // Fix initial conditions.
  prog.AddLinearConstraint( prog.initial_state().array() == x0.get_value().array() );

  // Fix final conditions.
  prog.AddLinearConstraint( prog.final_state().array() == xf.get_value().array() );
```
We have a plan for getting rid of the `.array()`.  This step will get rid of the `.get_value()` -- which appear often!

RussTedrake, 2017-03-28:
n.b., in the example above, x0 and xf come from e.g.
```
  SimpleCarState<double> x0, xf;

  x0.set_x(0.0);
  x0.set_y(1.0);
  x0.set_heading(M_PI_4);
  x0.set_velocity(15.0);  // m/s = ~ 33mph

  const double initial_duration = 30.0;  // seconds
  xf.set_x(x0.x() + initial_duration*x0.velocity());
  xf.set_y(0.0);
  xf.set_heading(0.0);
  xf.set_velocity(x0.velocity());
```

hongkai-dai, 2019-03-25:
@RussTedrake Is this feature still needed?

EricCousineau-TRI, 2019-03-25:
(I'd like to echo the "scary" bit - we shouldn't be duplicating Eigen / numpy / etc. unless we gain some meaningful features - peeps should just use the types from the linear algebra / array libraries!)

jwnimmer-tri, 2019-03-25:
I would propose a different solution here -- things like `final_state` or `get_continuous_state` should just return an Eigen (ref or block) directly, not a BasicVector.  Then we never need get_value, no matter what operations are about to happen.

sherm1, 2019-03-25:
Unfortunately neither `Context::get_continuous_state()` nor `Context::get_continuous_state_vector()` can return even a BasicVector, let alone a nice Eigen object. That's because our state is not contiguous in general.

RussTedrake, 2019-03-26:
Calling `get_value` or worse `CopyToVector` still happens all over the place -- not just trajectory optimization -- and continues to detract from usability.  I will happily support proposed solutions.

jwnimmer-tri, 2020-03-11:
Now that `port.Eval(context)` returns an Eigen block by default, what still remains as the next biggest pain point?

RussTedrake, 2020-03-12:
A quick search on `CopyToVector` still finds 100+ matches on master.  A quick scan gives me the impression that many of these involve directly accessing the state vector from the `Context`.

jwnimmer-tri, 2020-03-12:
I do see a few that are doing the copy in order to then use operator overloading, but most seem like the Copy is because the user actually wants the Eigen in order pass into an API that requires an Eigen, or because they want `.head(...)` or other APIs that only exist on Eigen.  I'd support offering Context to return a State view using Eigen::Block, but that is not this bug.

In other words, rather than adding more features to BasicVector, maybe we should just use Eigen directly instead, in more situations?  Maybe as part of #9171?

sherm1, 2020-03-12:
Not clear to me how that would work for Diagrams since their states are composites of LeafSystem states (and Eigen doesn't have a type comparable to VectorBase). Any Context might actually be a DiagramContext. But I don't think there is a practical distinction between BasicVector and an Eigen vector so we should be able to improve things once we get down to basics.

jwnimmer-tri, 2021-11-11:
Per #9171 and related, our intention is to demote `BasicVector` (and `VectorBase`, etc.) to implementation details that are eventually removed, and instead provide Eigen types directly when users access a context, etc.  As such, this feature request is "wontfix".

closed 2021-11-11 by jwnimmer-tri