RobotLocomotion/drake #5654: Allow access to the optimal active set from solvers like Gurobi

rdeits, 2017-03-28:
While we're adding to the MathematicalProgram wishlist, it would be nice to be able to read the active set of inequality constraints at optimality. This may be difficult to do in a generic way, since not all solvers will support this feature. 

This is useful for, e.g. the QP inside the Atlas controller, which relies on the active set changing relatively infrequently.

hongkai-dai, 2017-03-29:
@EricCousineau-TRI , we also talked about this during the meeting today. @rdeits we will add this feature in the second quarter.

hongkai-dai, 2019-01-25:
Now GurobiSolver can obtain `GurobiSolverDetails` from the result of calling `Solve()` function. The details doesn't include the active set yet, but it should be fairly straightforward to add it.