RobotLocomotion/drake #5681: Synchronize decision variables and optimal cost in `MathematicalProgram` with unsuccessful solve attempts

EricCousineau-TRI, 2017-03-31:
Following from the [newsgroup discussion](https://groups.google.com/a/tri.global/forum/?utm_medium=email&utm_source=footer#!msg/simulation-archive/GyRuf1o7Wkk/720B9FuyBAAJ), when an unsuccessful solve attempt is encountered, it would be preferable to synchronize the decision variables and the optimal cost with the `SolutionResult`.

The rationale is not to use these as indicators of overall failure, as the user should be using the `SolutionResult`, but rather provide more diagnostic information if necessary.

Example situations from Hongkai's email:

* Solved: Optimal cost and decision variables set accordingly
* Error: Optimal cost and decision variables set to NaN
* Convex Optimizations
  * Infeasible: Optimal cost is set to +Inf, decision variables set to NaN
  * Unbounded: Optimal cost is set to +Inf, decision variables set to NaN
* Non-Convex Optimizations, additional cases upon the Convex Optimization ones
  * Least Infeasible: Optimal cost and decision variables set according to the least infeasible solution

Care should be taken that this does not interfere with an implementation for warm-starting.

\cc @hongkai-dai @soonho-tri @RussTedrake @ggould-tri

hongkai-dai, 2017-03-31:
Thanks!

For Convex Optimizations, I would propose

-  Infeasible: Optimal cost is set to the cost of the dual problem if it is available, otherwise set to +Inf.  The decision variables are set to NaN.
- Unbounded: Optimal cost is set to -Inf (since we are minimizing the objective). Decision variables set to NaN.

hongkai-dai, 2019-03-26:
We have `kInfeasibleCost` and `kUnboundedCost` in MathematicalProgram to synchronize the cost

closed 2019-03-26 by hongkai-dai