RobotLocomotion/drake #5683: Provide support for warm-starting with the `MathematicalProgram` as constraints and costs change

EricCousineau-TRI, 2017-03-31:
Users of the `MathematicalProgram` should be able to warm-start a partially altered problem across different time steps.

Presently, users may set the initial guess manually, such as in [`inverseKinBackend`](https://github.com/RobotLocomotion/drake/blob/d5df8ec/drake/multibody/inverse_kinematics_backend.cc#L292); however, as evident in [line 218](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/inverse_kinematics_backend.cc#L218), the `MathematicalProgram`, as implemented, must be reconstructed at each time step.

This will also enable model predictive control to be more efficiently realized using the `MathematicalProgram`.

_[More details to be added]_

\cc @hongkai-dai @soonho-tri @RussTedrake @sammy-tri

hongkai-dai, 2017-03-31:
The goal is to allow model predictive control, where only a small set of constraints are changed, and the constraints being active in the previous time step, are likely to be active in the current time step. 

The current problems in `MathematicalProgram` include 
1. For most of the `Constraint` object, we do not have an `UpdateConstraint` function.
2. `MathematicalProgram` stores the constraints and costs, and then in `MathematicalProgramSolverInterface`, such as `GurobiSolver`, it parses ALL constraints and costs.  So updating one constraint requires `GurobiSolver` to reparse ALL constraints. On the other hand, gurobi provides the API to update each constraint individually (for example, see https://groups.google.com/forum/#!topic/gurobi/rkKUhsalZXw), we are not using this API, since we do not store the correspondence between our `Constraint` objects, and the parsed constraints in side gurobi solver (`GRBConstr` objects).
3. We do not return enough information for warm starting. We should return which constraints are active at the end of the solve.

jwnimmer-tri, 2019-03-26:
Duplicate of #4519.

closed 2019-03-26 by jwnimmer-tri