RobotLocomotion/drake #5693: snopt_solver.cc should not have dense gradient for the cost

hongkai-dai, 2017-04-01:
Now for a cost function that does not depend on ALL decision variables, we set the gradient cost to be dense (possibly non-zero gradient for ALL decision variables in the problem). We should only allow non-zero gradient for the decision variables inside the cost function.

RussTedrake, 2017-04-06:
This didn't read the way I expected.  #5666 already makes the fix to pass in autodiff matrices with derivatives allocated for only the relevant variables.  I thought you comment in reviewing #5666 was about using sparse matrices instead of dense matrices?

hongkai-dai, 2017-04-07:
I mean that in the cost function, `snopt_solver.cc` sets `iGfun` to be 1, and `jGvar` to be ALL variables in the entire optimization program, as in https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/snopt_solver.cc#L421-#L424
So if we have an optimization problem with variables `x` and `y`, and our cost function is `f(x) = 2 * x`. Although the cost function only takes `x` as the input argument, SNOPT still thinks the gradient for the cost function is dense, i.e., the gradient w.r.t `y` can be non-zero.

jwnimmer-tri, 2017-05-05:
All issues must have owners; feel free to reassign elsewhere as needed (or close), etc.

jwnimmer-tri, 2018-12-21:
@hongkai-dai I think perhaps the Fortran rewrite of snopt_solver has resolve this issue?

hongkai-dai, 2018-12-21:
Yes, this was resolved a while ago, when I added the cost_gradient_index to snopt_solver.cc

closed 2018-12-21 by hongkai-dai