RobotLocomotion/drake #5702: No Default Branches Specifies in On-Ramp RoadNetwork

liangfok, 2017-04-03:
I first noticed then when trying to run the `MaliputRailcar` on a on-ramp road network created by `MonolaneOnrampMerge`. It wouldn't proceed through the merge because no default branch was specified. I had to modify the `MaliputRailcar` to instead select the first "ongoing" branch.

I'm not sure if the above is actually a defect. Maybe it can simply be documented? (Or is it already documented and I simply missed it?)

jadecastro, 2017-04-03:
I noticed this too when using `MonolaneOnrampMerge` to build a piecewise polynomial using `RoadPath`. I couldn't seem to obtain an ongoing lane from both the `onramp1` and `pre5` lanes.  I had tried combinations of using `MakeGroup` and adding a new branch point with some difficulty.  At some point I was getting errors to the effect of the fact that the lanes I was attempting to specify were not confluent, which is true, since I created one of those against the intended direction of travel.

@maddog-tri - perhaps you could provide some guidance?

jadecastro, 2017-04-03:
FTR, this is the code in question: https://github.com/RobotLocomotion/drake/blob/master/drake/automotive/monolane_onramp_merge.cc

jadecastro, 2017-04-03:
Update: Looking back on the code and the default output produced above, I think the answer to the original question is that yes- we should include the default branch in that code. 

> At some point I was getting errors to the effect of the fact that the lanes I was attempting to specify were not confluent, which is true, since I created one of those against the intended direction of travel.

I see now that the reason for this might have been a silly mistake on relying on the graphic in #5169 that isn't in sync with the code. I will update that graphic and fix the branch point problem in a separate PR.

PR #5710 (2017-04-04): Fix ongoing lanes in MonolaneOnrampExample
resolves #5702.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5710)
closed 2017-04-04 by jadecastro