RobotLocomotion/drake #5704: village-demo broken on master?

RussTedrake, 2017-04-03:
after a clean build/install on xenial, running `bazel run drake/automotive:village-demo` resulted in 

```
...
[dev_demo] Welding joint front_lidar_joint
[dev_demo] Welding joint top_lidar_joint
[dev_demo] Welding joint rear_right_lidar_joint
[dev_demo] Welding joint rear_left_lidar_joint
[dev_demo] terminate called after throwing an instance of '
[dev_demo] std::runtime_error
[dev_demo] '
[dev_demo]   what():  
[dev_demo] CarVisApplicator::DoCalcOutput(): Input PoseBundle has 0poses. Expected 15.
[lcm-logger] Summary: lcmlog-2017-04-03.00 ti:   2sec Events: 5         (    0 MB )      TPS:     2.73       KB/s:    30.00
[automotive_demo.py] dev_demo exited -6
```

apologies in advance if it's user error!

RussTedrake, 2017-04-03:
not surprising, but the merging demo has the same problem.

liangfok, 2017-04-03:
I know what the problem is. `AutomotiveSimulator<T>::AddPriusEndlessRoadCar()` is not connecting the `EndlessRoadCar` to the `PoseAggregator`. I shall self-assign and fix this with high priority.

liangfok, 2017-04-03:
Maybe a better fix for this is to replace the `EndlessRoadCar` with `MaliputRailcar`? There are still some kinks to work out regarding `MaliputRailcar`'s lane transitions. They are being worked out in #5699. I'll see which one gets into master first.

jwnimmer-tri, 2017-04-03:
I am going to `git rm` the village / endless code now.  If we want to make a video of it, we should use a commit from when it went in.

PR #5705 (2017-04-03): Remove the infinite / endless demo
Closes #5704.

This demo code's use of the "InifniteRoad" concept is inconsistent with the modeling choices we want to use moving forward, and our new railcar supplants the "endless" car.  I'd been hoping to get the new highway merge ready before removing this, but I guess the lack of testing here means its bitrotted anyway, so might as well just remove it now.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5705)
closed 2017-04-03 by jwnimmer-tri